Drewbadour/Xone-OSX #16: Control mapping incorrect in Shovel Knight / Steam

keithchu, 2015-01-02:
I tried going through Big Picture Mode and remapping the controls, but nothing I changed in the config had an impact on the controls that registered in-game.

From what I can tell, in Shovel Knight, this how the Xbox One controls are mapped. The left value is what was pressed on the controller, the right value is what the game interpreted the button press as. The stick movements and trigger presses were interpreted perfectly.

X = Select
Y = A
A = <no input>
B = Start
LB = B
RB = X

Select = Y
Start = D-pad / Up
 
D-pad / Up = RB
D-pad / Down = L-Stick click
D-pad / Left = Select
D-pad / Right = <no input>

L-Stick click = D-pad / Down
R-Stick click = D-pad / Left

Do you know of any way I can remap the controller just for this game? Since Shovel Knight is pretty simple, I can just remap most of the inputs to the correct controller button. The main issue here is that the "A" button isn't mappable and can't be set to Jump.

FranticRain, 2015-01-02:
Shovel Knight chose a very odd route for their controller support. There is a way you can send them your controller mapping and they add it in the next update. But this new version of the driver I am writing that acts like the 360 controller should do that for you. I actually sent in a mapping early on, but it was the Rev 1 mapping and not the current mapping, nor is it the future mapping. Just give me a couple days to round all this up, and then you should be able to enjoy the shovelry.

keithchu, 2015-01-02:
Fantastic – thanks for the quick reply! Looking forward to the new version :)

FranticRain, 2015-01-03:
Okay, I updated the driver a bit faster than I thought it would take. My version of Shovel Knight is the GoG version, so make sure it works with the Steam version. Report back so I can tell if I need to close the issue.

keithchu, 2015-01-03:
It works perfectly! Thanks @FranticRain!

closed 2015-01-03 by keithchu