Drewbadour/Xone-OSX #19: Controls in preference pane XIB are hidden?

pizthewiz, 2015-01-03:
I was looking to make some tweaks to the preference pane and I can't quite figure out why none of the controls seem to be visible:
[image]

It is as if the controls are not in the selected tab, though given the hierarchy, they most definitely are. I think something about the configuration makes Xcode 6.1.1 and 6.2 Beta 3 extremely sluggish, perhaps related to the controls being hidden?

FranticRain, 2015-01-03:
You have to actually click on the "general" button to make them show up. Its super glitchy and lags like crazy no matter if they are hidden or not. It just really hates images for whatever reason. I usually edit it in code.

FranticRain, 2015-01-03:
Another method is to go to the NSTabView and set the tab total to 2, and then all of the images will magically appear on the first tab. Still is a laggy mess though.

pizthewiz, 2015-01-03:
Super strange, I could only see the content by adding a tab then removing it - filed it as rdar://problem/19369003 in case they can sort it out at some point, closing this out…

closed 2015-01-03 by pizthewiz