Drewbadour/Xone-OSX #20: Doesn't Work With Cook Serve Delicious

bhelyer, 2015-01-05:
Doesn't react to any input. The underlying engine (GameMaker) is the same as Risk of Rain, no idea if that's related. Interesting thing is, it has Xbox One controller support (supposedly) -- I've not uninstalled the driver to see if it works, though.

I asked on their forums, too: http://steamcommunity.com/app/247020/discussions/0/627456486569628697/

FranticRain, 2015-01-05:
It actually does send input. If you put it into Xbox 360 mode and then press the right stick it will move the D-pad right, but of course that's not a good option at all. I did take the liberty of uninstalling the driver and trying it out. It did absolutely nothing. Hopefully I can work with the CSD dev to get GameMaker games supported. I'll post in that thread so he can get in contact.

Thanks for the find, getting a dev's help on getting compatibility working is going to be awesome!

FranticRain, 2015-01-05:
As an aside, I can't get it to work with a wireless 360 controller using the https://github.com/d235j/360Controller driver. I think GameMaker has some sort of gamepad issue with Yosemite.