Drewbadour/Xone-OSX #21: Does not recognize Rock Candy Xbox One controller

J-Fricke, 2015-01-05:
Controller is Officially Licensed, so it should fit the same spec as the MS one.

I wonder if this just means it's missing productID and vendorID entries or something like that?

J-Fricke, 2015-01-05:
I believe this is the Hex #s for this device: VID_0E6F & PID_0146

Found here: HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Enum\USB\VID_0E6F&PID_0146

I can track down more info if needed.

FranticRain, 2015-01-06:
Thanks for the comprehensive information. Let me know if v0.4.1 solves your problem.

J-Fricke, 2015-01-06:
Not working, though I looked at the code here: https://github.com/FranticRain/Xone-OSX/blob/master/Xbox%20One%20Controller%20Driver/Xbox%20One%20Controller%20Driver/Info.plist
And I believe you have product and vendor #s swapped.
vendorID / idVendor = 0E6F (hex) or 3695 (decimal)
productID / idProduct = 0146 (hex) or 326 (decimal)

FranticRain, 2015-01-06:
Ugh. I feel super stupid for messing that up, sorry. I fixed it, so download the release again and hopefully it works this time. Sorry about that.

J-Fricke, 2015-01-06:
Ok now it is recognized in Steam as X360Controller, however in the PrefPane it does not list a controller in the dropdown at all and I cannot get any binds to work in steam either. 

I have Joystick Mapper app as well and it sees a joystick though when I hit scan, it also does not see my button, stick or d-pad presses/movement.

Also no lights light up (xbox symbol) when plugging in nor does it rumble, this might just be missing features though it is what I see/feel when plugging into windows 7.

Let me know how I can help further on this.

FranticRain, 2015-01-06:
So since its showing up in Steam, it sounds like the kext is loaded. What happens when it is loaded is that the controller is sent 0x05, 0x20. The packet that is sent by the Windows driver sends a slightly longer packet that causes a rumble. If there is some way for you to sniff the packet that Windows sends your controller, then I can use that to boot the controller. I don't know if its different for your controller, or if the longer packet is what makes third party controllers work.

geowar1, 2015-01-07:
If I may make a request... Can you make the 360 mapping an option in the pref pane? I really only care about the Xbox one controller.

FranticRain, 2015-01-07:
In what way? Do you want an option to have the controller change its product id and product string in the preference pane? Because as far as I know I can't have multiple HID report descriptors.

geowar1, 2015-01-07:
I don't care/need the 360 mappings. You can change what product/vendor IDs are returned by the driver at initialazation time. It would then just require the user to un-plug and re-plug the controller to get the other mapping.

FranticRain, 2015-01-07:
How would you want the buttons laid out in this alternate version? Mapped to the same button numbers as the 360 controller, but without the re-organization of the packet?

J-Fricke, 2015-01-07:
I tried the one above but it gives an error and is a smaller file that the standard .pkg file. 641 vs 37kb

J-Fricke, 2015-01-07:
The operation couldn’t be completed. (com.apple.installer.pagecontroller error -1.)

FranticRain, 2015-01-07:
Curious. I downloaded the driver from that link and just installed it myself. It worked just fine. Its smaller because I removed some excess garbage from the kext. Not sure how to proceed.

FranticRain, 2015-01-07:
Lets try a manual install then. [Download the kext here](https://mega.co.nz/#!u0AXgYAD!C2Qs0Lwwt6zBWmq7rtjFoDMLuOOV7xhXShsMiruel4A) and then use this command on the kext in the directory you downloaded it to. For example if you put it to your downloads,
```sudo chown -R root:wheel "~/Downloads/Xone Driver.kext"```
And then move it into your /Library/Extensions folder the way you like, but you could simply do:
```sudo cp -R "~/Download/Xone Driver.kext" "/Library/Extensions"```

geowar1, 2015-01-07:
>> How would you want the buttons laid out in this alternate version? 
Your original Xbox one ReportDescriptor[253] in xonehid.h was correct (if overly verbose).

FranticRain, 2015-01-07:
Okay. I'll return this driver to that report descriptor after I get the 360-faking version integrated into the revived Tattiebogle driver.

geowar1, 2015-01-07:
J-Fricke: if after the manual install it still doesn't work you can use kextutil to figure out why:
`sudo  kextutil -t "/Library/Extensions/Xone Driver.kext"`
post any errors back to this forum.

J-Fricke, 2015-01-08:
Manually it worked, unsure why that .pkg was not working. I had to remap some buttons in steam to match the controller as the a-b were reversed it seems.

FranticRain, 2015-01-08:
I'll keep looking into this. That A-B swap doesn't make any sense.

FranticRain, 2015-02-21:
Does this new 1.0.0 Release Candidate do anything for your problem? I don't think that it will, but it might be worth a try.

esaumusic, 2015-05-21:
Just found your driver/utility and installed latest 1.0.3. Works great with official x1 controller, so stoked, thanks sooo much!  Still unsuccessful with my Rock Candy controller, thought I'd let you know. 

Here's what my system info says:

Rock Candy Wired Controller for Xbox One:

  Product ID:	0x0146
  Vendor ID:	0x0e6f
  Version:	1.01
  Serial Number:	0000ACC77A6CBFB5
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0xfa130000 / 6
  Current Available (mA):	500
  Current Required (mA):	Unknown (Device has not been configured)

FranticRain, 2015-05-21:
[Please download this version of the driver](https://dl.dropboxusercontent.com/u/33409904/Xone-Driver-1.0.3-r3.dmg) and then plug your controller in and check your Console.app logs. There should be an entry that says "Score - XXXXX." For example, with the official controller, it should say "Score - 90000." Let me know if you get a score and what score you get.

esaumusic, 2015-05-21:
Ok installed new version, restarted, opened console but no messages when I plug in the rock candy. I do see the 90000 score for my official controller like you said. It reads:

5/21/15 12:42:55.000 AM kernel[0]: Score - 90000
5/21/15 12:42:55.000 AM kernel[0]: Xbox One Controller - handleStart

But nothing for rock candy. FYI, it does work on my main gaming desktop pc. But sometimes I'm on the road with the macbook and need a quick gaming fix ya know? ;) Thanks for taking a look!

FranticRain, 2015-05-29:
Please uninstall my driver and try using this beta version of this driver instead: https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview7.pkg

I've been porting my code into it, and it definitely works with the Microsoft controller. Hopefully it also works with third party controllers.

esaumusic, 2015-05-30:
Nice man, rock candy lights up and is working! Tested it on a few indie games and some of the buttons seem different than they were on my pc but mostly seems to work fine. Also, where's your donation link? :)

FranticRain, 2015-05-30:
I don't have a donation link. I'm just glad it works for you. If you aren't happy with the way the buttons are bound, that driver has a button rebinding functionality built into it. Its a driver level change though. Its not per game. In the future, continue to download [the 360Controller repsitory](https://github.com/d235j/360Controller) because this one will be deprecated. If you have any futher problems, I'll be around there as well as here.

closed 2015-05-30 by FranticRain

HVSNick, 2015-10-02:
Hey all, I know this is marked as closed, but I cant seem to get it to work. I've tried every version of the drivers and tried the special one above a few times with no luck. I've also tried multiple Xbox One Rock Candy controllers with multiple cords. Nothing. 

Any ideas? Feel like I'm going crazy

FranticRain, 2015-10-02:
Uninstall this driver first. You can check that by making sure the preference pane is gone. Additionally, there shouldn't be a "Xone Driver.kext" in /Library/Extensions. Then, [install the latest version of this driver.](https://github.com/360Controller/360Controller/releases) It should work with the Rock Candy controller.

HVSNick, 2015-10-02:
Sadly doesn't work. Tried that one too. Also just tried a fresh install (after removing all Xbox drivers from the Extensions folder). :(

FranticRain, 2015-10-02:
Have you tried follow the debugging steps listed in the README of this driver? Does the controller appear in the System Report when plugged in?

HVSNick, 2015-10-02:
I just read it but it's over my head. Sadly I'm just a lowly Producer that barely knows code stuff at a game studio. We have a whole crate of these things for our staff, half of which are on iMacs.

FranticRain, 2015-10-02:
Its really simple, there are pictures and everything. Press the Apple button -> About This Mac -> Overview -> System Report -> USB. There should be a list there. It should contain an object called "Controller" If it is there, let me know the Product and Vendor IDs that are listed when you click on it.

HVSNick, 2015-10-02:
Boom thanks. May have been looking at the wrong Read Me. I have two plugged in that weirdly have different version numbers:

Product ID:	0x0246
  Vendor ID:	0x0e6f
  Version:	1.01
  Serial Number:	000068864143CDA4
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0xfa120000 / 5
  Current Available (mA):	500
  Current Required (mA):	Unknown (Device has not been configured)

-----

  Product ID:	0x0246
  Vendor ID:	0x0e6f
  Version:	1.10
  Serial Number:	000077A7F0A6C23D
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0xfd130000 / 5
  Current Available (mA):	500
  Current Required (mA):	Unknown (Device has not been configured)

HVSNick, 2015-10-02:
Looks like the product ID has ticked up  from 146 to 246 since this thread began.

FranticRain, 2015-10-02:
Oh, let me guess. It has a 3.5mm jack on the bottom of it now, doesn't it?

HVSNick, 2015-10-02:
Actually it doesnt!

FranticRain, 2015-10-02:
Well thanks a lot PDP! Change the product ID and don't actually update the controller. That's just rad. I'll put in a pull request for you with that new product ID on the 360Controller repository. You'll have to wait until someone else can do a new build for you though. Once that version is out, everything should work perfectly. So just keep your eyes over there for now.

FranticRain, 2015-10-02:
[You can find the pull request here.](https://github.com/360Controller/360Controller/pull/151) If you post in there, Rodrigo will make sure to let you know when he's built it.