Drewbadour/Xone-OSX #22: Does not work with Aspyr-ported titles

kram530, 2015-01-07:
Aspyr is the company responsible for a lot of big name ports from pc to mac fairly recently (Borderlands 2 and Bioshock Infinite are the ones I own and can't get working).  Aspyr has said that controller functionality is working for only xbox 360 controllers through drivers that are installed in the game itself apparently.  I was hoping to play both these games with my xbox one controller as I don't own a 360 controller that I can use.  While I do have a keymapping utility that will allow me to set the controller inputs to input keystrokes, I've never been able to get that type of workaround to go smoothly.  I'm pretty certain that a fix for one of these supported Aspyr-ported release titles will work across them all.  The list of controller supported aspyr games can be found here: 
http://support.aspyr.com/hc/en-us/articles/203396034-Aspyr-Games-Controller-Support-and-you-
I have loved this driver and watching it develop has been awesome so far.  I just came back for a patch the other day and it fixed support for configuring the controller in the Binding of Isaac: Rebirth just as I was hoping for.  Keep up the great work!

FranticRain, 2015-01-07:
Unfortunately, I don't think I'm going to be able to get it to work with Aspyr games. That page is pretty explicit on the no-third-party-drivers front. I've got a pretty good idea of how it is most likely implemented and at this point I don't think I can fix it. My only idea is to create a separate kext that exists for the sole purpose of making the Xbox One controller look like and Xbox 360 controller and then altering the messages back and forth. This is another classic example of why Apple's implementation of controllers is a nightmare.

So I can't really promise anything, but I'll keep an eye on it.

kram530, 2015-01-07:
Okay, thanks for looking at it.  I'm pretty annoyed with the difficulty of implementing controllers on mac in general.  It seems like it should be so much simpler.  Hopefully the future will lead to simpler times on this front.  Thanks again for all your awesome work!

FranticRain, 2015-01-26:
I have been working to integrate Xbox One controller support into a different driver that plays nicely with Aspyr ported titles. Now note that this is an early release of a driver, and you should do your best to get a proper release as soon as they officially put out the 0.15 update. But this will solve your problems for now.

First, make sure you use the uninstaller included in the latest release, not a pre-release. Next, download and install [this driver](https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview2.pkg). And then from then on, pay attention to [this repository](https://github.com/d235j/360Controller/) for future updates to that driver. This should work perfectly. Good luck!

smoketetsu, 2015-03-20:
Since that's a x360 driver I tried testing it with my x360 controller  and for me both the 0.12 driver and that work fine with either Borderlands 2, Pre-sequel and BioShock Infinite. Albeit without rumble. 

Neither driver works with Star Wars The Force Unleashed though which I consider a holy grail for getting working with controllers that need third party drivers.

FranticRain, 2015-03-20:
Since this is out of my control and in Aspyr's control, you'll have to use the pre-release version of the 360Controller driver. I see you are familiar with it, and there are links in my previous post.

smoketetsu, 2015-03-20:
Yeah that's why I was reporting in.. because I just tested that driver you linked to in your previous post. It still doesn't work with SWTFU. There's also a problem with rumble in newly enabled rumble games but I mentioned that in the repo for the driver itself (like for example DMC4 in Wine, or Distance in Steam) when a rumble event happens it seems to keep on going until another rumble event interrupts it.

FranticRain, 2015-03-20:
Unfortunately, the SWTFU issue will be impossible to fix. That particular game ignores any sort of drivers that OS X provides, and uses direct input from a wired Xbox 360 controller. So it is impossible for us to fake this hardware interaction. In addition, the force feedback plugin is not my area of expertise, so this isn't the best place to ask.

smoketetsu, 2015-03-20:
Whose area of expertise is it? Someone at the 360controller repo? I left a comment about that there as well.

A shame about SWTFU. Especially since native controller support would be the best and keyboard emulation would be a sub-par solution IMO.

FranticRain, 2015-03-20:
MaddTheSane wrote the current incarnation of the force feedback driver. If you give him a bit of time, I'm sure he will be able to help you with your force feedback related problems.

smoketetsu, 2015-03-20:
I just noticed Distance from Steam is now hanging on load with that preview driver. I have to force quit it.

FranticRain, 2015-03-20:
Delete the files located at:
```/System/Library/Extensions/360Controller.kext
/Library/Extensions/360Controller.kext"
/Library/Extensions/Wireless360Controller.kext
/Library/Extensions/WirelessGamingReceiver.kext```
And then fully reboot your computer. This will tell you if it is the driver's fault or not. It wouldn't make any sense that it causes a problem since the driver is loaded when the controller is connected to the machine, and the method by which the driver is shown to the application is not out of the ordinary.

smoketetsu, 2015-03-20:
Well the game worked just fine with the last official 0.12 drivers from tattiebogle's site but not this driver. So I don't know. I'll try out some stuff and then report back in.

liamorzen99, 2015-11-05:
I know this is very late in the convo but is there a Xone driver that works for borderlands 2 haven't been able to get it working hope you can help.

FranticRain, 2015-11-05:
You best bet is [this driver.](https://github.com/360Controller/360Controller/releases) If it doesn't work with that one, it likely won't ever work.

liamorzen99, 2015-11-05:
Ill give it a try. Thank You

liamorzen99, 2015-11-05:
Ok so the driver show that my Xbox One controller is there but doesn't work on borderlands is there something else i have to do to get it to work?

FranticRain, 2015-11-05:
Nope. If it doesn't work, it doesn't work.

liamorzen99, 2015-11-05:
Ok well thanks anyway.