Drewbadour/Xone-OSX #23: Doesn't work on 10.10.2

horsetamer, 2015-01-11:
Hey, I just installed your driver and attempted to plug in my controller. The batteries were not in it, and the light would not come one. I pressed the home button and the light began to blink which indicates there is power from the micro-USB to USB connection. Thanks for any help. Also I am unsure of how to uninstall this driver.

FranticRain, 2015-01-11:
To uninstall the driver, all you have to do is go to /Library/Extensions and remove the file "Xone Driver.kext". But before you do, I'd like to see what the problem is. If you could, could you run `sudo kextutil -t "/Library/Extensions/Xone Driver.kext"`from your terminal so I can figure out what the problem is?

horsetamer, 2015-01-11:
new-host:~ horsetamer$ sudo kextutil -t "/Library/Extensions/Xone Driver.kext"
new-host:~ horsetamer$ 

This is what I get from running that

FranticRain, 2015-01-11:
Did you try plugging in the controller after you ran that? It should work. Make sure to run it again before you test. It'll unload do to inactivity.

horsetamer, 2015-01-11:
It's the same issue, nothing happens

FranticRain, 2015-01-11:
Is this a Microsoft Xbox One controller? Could you check the Product ID and Vendor ID of the controller under: Apple Menu -> About This Mac -> System Report -> Hardware -> USB. The device is likely called "Controller" under that list.

horsetamer, 2015-01-11:
Yes it is a Microsoft Xbox One controller, it's this one:

http://www.bestbuy.com/site/microsoft-xbox-one-special-edition-midnight-forces-wireless-controller-camouflage/7307001.p?id=1219265949449&skuId=7307001

On my USB Device Tree the controller actually doesn't seem to show up, I'm pretty sure the micro-USB is making a connection since it will light up when I press the xbox button and there are no batteries in it. I'll go grab my other xbox controller and another micro-usb cable and check this out.

Thanks for all your help so far man!

FranticRain, 2015-01-11:
I was afraid that those camo controllers would have different Product ID's. You can also check that number in Windows if you have that handy.

horsetamer, 2015-01-11:
Same issue with the black controller and a different micro-USB, neither of the devices seem to show up in my USB Device Tree

FranticRain, 2015-01-11:
Well thats interesting. Thats some core incompatibility with 10.10.2. If the OS can't even see it, then my kext definitely isn't going to. This seems like its out of my hands, and is Apple's fault. I guess you just have to report that OS X can't see your controllers to Apple and hope they fix it. Also consider uninstalling the driver `sudo rm -rf "/Library/Extensions/Xone Driver.kext"` and then checking the device tree again. This is such a weird bug.

horsetamer, 2015-01-11:
Yeah this is kinda weird, thanks for all the help! I'll definitely let you know if I figure this out.

geowar1, 2015-01-11:
That is more likely a charging cable not a full USB signal cable.

horsetamer, 2015-01-11:
I didn't know something like that existed, I'll look around for a different cable

horsetamer, 2015-01-11:
Wow, props to you geowar1, I tried a different cable and it completely worked this time.

FranticRain, 2015-01-11:
Totally didn't know that they made charge-only cables. At least its solved now!

closed 2015-01-11 by FranticRain