Drewbadour/Xone-OSX #24: Controller not seen

bkchucks, 2015-01-14:
I have the preference panel open and everything installed. When I plug my controller and press the home  button it just flashes. The preference panel does not show me a list of controllers, just empty spots. I know my USB is powering my controller without batteries, but it does not connect to my computer. 

I am on the latest Yosemite.

FranticRain, 2015-01-14:
The preference pane only picks up the controller after it has been plugged in, so don't worry about it for now. I need to see if OS X is actually recognizing that a controller is connected. Check: Apple Menu -> About This Mac -> System Report -> Hardware -> USB. Do you have a device is called "Controller" on that list?

bmfs, 2015-02-09:
Same thing happens to me. 
The Controller is visible from the System Report Window.
I'm using OS X 10.9.2

FranticRain, 2015-02-09:
Does the controller light up at all? Is it a Microsoft made Xbox One controller or some third party controller?

bkchucks, 2015-02-10:
So the problem is that the cable I'm using only supports power transfer and not data. Find the right cable and everything works.

FranticRain, 2015-02-10:
Correct. A controller plugged in via a charge cable will not appear in the system report, though.

Aviatorgamer, 2015-04-25:
SO. I was using my controller great for a while, but I was having issues in Black Ops not seeing the controller at all because it's made exclusively for 360. I tried a different xbox one controller solution and uninstalled this one. with no success, I uninstalled the other one and re-installed this one. Now, I cannot use my controller in any game at all because none of them recognize that the controller is plugged in. the Syspref pane shows the controller and responds to input and the controller's light is solid, but games are completely unresponsive to the controller.

OS X Yosemite 10.10.2
Macbook Pro 13" 2015

FranticRain, 2015-04-25:
Did you download the -360 version of the driver? You need that one to get compatibility with games.

In the future, please create a new issue from problems such as this.

Aviatorgamer, 2015-04-25:
No I did not try that, but previous to my issue, I was able to play games flawlessly without any additional drivers other than this one

FranticRain, 2015-04-25:
There are two versions of the driver. The Xone-Driver-1.0.1.dmg and the Xone-Driver-360-1.0.1.dmg. The former was requested by a few people to be bound to controls similar to the Xbox One. The latter changes the controller to act like an Xbox 360 controller. This is the one you want.

Aviatorgamer, 2015-04-25:
This did fix my issue, Thank you! Is there any way to figure out how to trick a game (in my case, an Aspyr game) into thinking that this controller is actually a 360 controller. This would make compatibility better in games that only support the 360 controller exclusively

FranticRain, 2015-04-25:
Unfortunately, Aspyr bypasses any sort of driver in their games. They access controllers directly. There is absolutely nothing any driver can do about that.

Aviatorgamer, 2015-04-25:
Okay. Thank you for all your help!

tcook1231, 2015-07-30:
I am using a xbox one arcade stick. It is being seen by the OS but not the driver. It uses all the same inputs as a xbox one controller. 
The product and Vendor Ids are the following
Product ID:	0x0063
Vendor ID:	0x0f0d  (HORI CO., LTD.)

FranticRain, 2015-07-31:
@tcook1231 In the future, please create a new issue rather than tagging on to an unrelated one. Unfortunately, my driver only works with Microsoft controllers. However, I have been moving my code into a driver that does support third party controllers. Please uninstall my driver and install [this one instead.](https://github.com/d235j/360Controller/releases/tag/v0.15_beta_unofficial) If that doesn't work, let me know.

tcook1231, 2015-07-31:
sorry about that I saw this and just posted.

FranticRain, 2015-07-31:
It's fine, just let me know if that solves your problem or not.

tcook1231, 2015-07-31:
It isn't I am posting on that issue page.