Drewbadour/Xone-OSX #25: Controller unresponsive after machine sleep/wake

pizthewiz, 2015-01-15:
While the device can be seen via _USB Prober.app_ or _System Information.app_, on machine wake the Home/Xbox button does not re-illuminate and the none of the controller's inputs seem active in the pref pane. An un/re-plug does the trick, but I wasn't sure if there was something else that could be done.

FranticRain, 2015-01-15:
How long does the computer need to be asleep see this happen? I've been messing with it and the controller just stays on when I sleep my laptop.

pizthewiz, 2015-01-15:
For me, it took perhaps 60 seconds for the Home/Xbox button to go out.

FranticRain, 2015-01-16:
Try this pre-release. Don't know if it'll work or not. https://github.com/FranticRain/Xone-OSX/releases/tag/v0.4.2

pizthewiz, 2015-01-18:
Sadly no dice, this is occurring on a `MacBookPro11,3` running 10.10.2 and occurs when plugged or not. I'll try it on an `iMac11,1` and see if I have better luck over there.

pizthewiz, 2015-01-18:
The controller seems to stay powered up through sleep of my `iMac11,1`, though the machine seems to have power management issues so I'm not sure it is a great reference point.

FranticRain, 2015-01-19:
I figured out I had NoSleep installed and thats why my controller wasn't going to sleep. So now I can finally test this and fix it. It seems I'm not getting the "kIOMessageDeviceSignaledWakeup" message. It's unwillingness to appear may be related to that renegade "kIOMessageServiceIsRequestingClose" that shows up when the driver starts. I'll keep working on it.

FranticRain, 2015-02-18:
[Latest release candidate should fix this issue.](https://github.com/FranticRain/Xone-OSX/releases/tag/v1.0.0-rc) Let me know how it works.

pizthewiz, 2015-02-18:
Whoa, that is a substantial commit - I'll give this a try soon.

pizthewiz, 2015-02-21:
Yup this now works! The installers lost their signatures but this is good to go.

closed 2015-02-21 by pizthewiz

FranticRain, 2015-02-21:
D'oh. Can't believe I forgot that. I re-uploaded the dmgs with signed pkgs. Should be fixed.