Drewbadour/Xone-OSX #28: Steam: Key Config

VirtualPenman, 2015-01-22:
The driver is great and I must say you have done a great job. I am having a problem on steam where it fully recognizes the controller while I'm in the steam app just controlling my library, settings, etc. But once I go to a game, (The binding of Isaac rebirth) the game says the controller is unrecognized and that I need to create a Key Config for it. Is there anything to do?

Thanks Frantic!

FranticRain, 2015-01-22:
Unfortunately, I don't know for sure what is causing this issue. However, I've been working to integrate the support into a more fully featured driver. If you are willing to install a beta version of a driver, which you must be if you are using this one, install [this driver located here.](https://github.com/d235j/360Controller/issues/28#issuecomment-70903592) You will also need to uninstall my driver. Do so by deleting /Library/Extensions/Xone-Driver.kext or by running the command `sudo rm -rf "/Library/Extensions/Xone Driver.kext"` in the terminal, you can delete my driver. You may also want to consider deleting my preference pane is it won't do you much good. But I believe that this driver should work perfectly with Steam and BoI:R.

VirtualPenman, 2015-01-22:
Hmm I tried this driver. Seems to be the same issue. Thanks for the help though!

FranticRain, 2015-01-22:
Have you tried rebinding the controls from within the application? Because if it doesn't support that driver, then it has no native config for the 360 controller on OS X.

VirtualPenman, 2015-01-23:
Do you mean within the steam app? I maped out all the controls from within steam but once I get into the game it says there is no key config

FranticRain, 2015-01-23:
No, I mean configure it in the game. [Like this](https://gaming.stackexchange.com/questions/190814/how-do-i-rebind-the-controls-in-boi-rebirth).

VirtualPenman, 2015-01-23:
I hadn't even realized that was a thing. This worked perfectly thank you for the help!

closed 2015-01-24 by FranticRain