Drewbadour/Xone-OSX #29: Questions about your code...

TheSafo, 2015-01-22:
Hi,
I'm trying to create a program to emulate an Xbox controller, so my program can interface with other games on the computer as a gamepad. Is this something I can do with your codebase. If not do you have any resources/suggestions on how I can go about doing this.

Thanks!

FranticRain, 2015-01-22:
Unfortunately, I don't know of anything. What you are going to want to look for is some kind of Human Interface Device emulation. I don't recommend you do any of this on OS X because quite frankly Apple's documentation on this stuff is garbage and it isn't a fun experience. But if I can't dissuade you, you might also want to check out [CGEventSetFlags](https://developer.apple.com/library/mac/documentation/Carbon/Reference/QuartzEventServicesRef/index.html#//apple_ref/c/func/CGEventSetFlags). You may be able to set flags that look like controller motions. Good luck.

TheSafo, 2015-01-22:
Thank you for the help! I'll look into CGEventSetFlags, in addition to looking into just doing a windows program. Thanks again for the quick response, and good luck!

closed 2015-01-22 by TheSafo