Drewbadour/Xone-OSX #31: Not working for Borderlands 2

zeGamelord, 2015-01-26:
Doesn't work right out the box for borderlands 2. I might not be doing something I'm meant to but I have no idea what that may be.

I've installed the controller and it works well from what I can see in system preferences.

FranticRain, 2015-01-26:
Unfortunately, Aspyr titles such as Borderlands 2 don't play nicely with my driver. However, I have been working to integrate Xbox One controller support into a driver that does play nicely.  Now note that this is an early release of a driver, and you should do your best to get a proper release as soon as they officially put out the 0.15 update. But this will solve your problems for now.

First, make sure you use the uninstaller included in the latest release, not a pre-release. Next, download and install [this driver](https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview2.pkg). And then from then on, pay attention to [this repository](https://github.com/d235j/360Controller/) for future updates to that driver. This should work perfectly. Good luck!

zeGamelord, 2015-01-26:
I just tried the driver out.
Still not working, the driver recognises the controller but Borderlands 2 is entirely oblivious of it's existence.

FranticRain, 2015-01-26:
Okay so lets go over the basic checklist. First, Reboot, just in case. Next, in the preference pane, can you interact with the controller and it shows up? Is the xbox button lit up? When you start the game, does the controller menu show up? I tested Borderlands 2 with my controller today and it worked just fine. Do you have any other games that the controller is working with? Also, is your version of Borderlands 2 from the Mac App Store or on Steam? Mine is the Steam version.

dvillani, 2015-02-08:
I have the xone controller and it does not work with Borderlands 2 from the mac app store any ideas towards why?

FranticRain, 2015-02-08:
Borderlands 2 for Steam and the Mac App store are different beasts altogether. Aspyr felt the need to put controller support in themselves, so they don't support drivers very well. The steam version appears to have some sort of override for the Tattiebogle driver for the Xbox 360 controller. I've been working on putting the Xbox One controller support into it, and I know that it works with the steam version. [Check it out.](https://github.com/d235j/360Controller/issues/28#issuecomment-72572476) Make sure you uninstall my driver first.

dvillani, 2015-02-08:
works perfectly now, thanks for the immediate feedback, nice work

closed 2015-02-08 by FranticRain