Drewbadour/Xone-OSX #32: Futuridium EP Kern/Crash

s0nera, 2015-01-29:
Futuridium EP is a free game in the App Store, not steam.
After I installed the driver, the game started to crash at launch. I don't know if the driver is related to it, but  it had never happened before.

I tried uninstalling the driver, safe booting, reverting the kext signature checking, but the geme seems not to work anymore.

I looked for help googling the error and inside mac forums, but no success.

Could you please help?

FranticRain, 2015-01-29:
I'm trying to recreate this issue but I can't. Are you getting a kernel panic or is the game just crashing? Make sure that there isn't a "Xone Driver.kext" in /Library/Extensions. Only other thing I can recommend is reinstalling Futuridium.

s0nera, 2015-01-29:
Never mind. I logged as the guest user and the game worked. It has nothing
to do with the driver. Sorry bothering you guys.

FranticRain, 2015-01-29:
Don't worry about it. Good luck in the future.

closed 2015-01-29 by FranticRain

s0nera, 2015-02-08:
Well. Due to my lack of knowledge and some other performance issues, I ended formating my mac. Now the driver and the game works, but the buttons are not right and the game has no controller mapping editor or anything like that besides inverting Y axis. also, should I open a new issue?

s0nera, 2015-02-08:
Never mind (again) I just found the control map editor. I'm trying to deal with my lazy way to deal with problems, like asking people before trying to solve for myself.
Sorry and thanks.

FranticRain, 2015-02-08:
Hahaha, not a big deal. Don't worry. Ironically, I just added a remapping function to [this driver.](https://github.com/d235j/360Controller) I've been adding Xbox One controller support to it. So when they release 0.15 you could try that alternately. Happy that its working out for you.

s0nera, 2015-02-09:
Just tried it this weekend. Awesome work! Both 360's and One's controller
working at the same time. Now I don't have enough USB ports for my friends
to play with me.