Drewbadour/Xone-OSX #34: If this is an application it won't install.

Scorpian4875, 2015-01-31:
I downloaded the master folder with all the files and installed the .pkg file, yet when I restart my computer there are no new files or applications that I would've expected to be installed. I'm not sure what the problem is...

FranticRain, 2015-01-31:
If you open up the System Preferences, is there a new preference pane or is it as it was? The installer doesn't actually install anything into your Applications folder. The driver itself is actually located in /Library/Extensions.

Scorpian4875, 2015-01-31:
Ah that would make sense, I actually just left my house but I'll check when I get home. Thanks for the quick response!

Scorpian4875, 2015-01-31:
I just checked and there is a new option in settings with all the controller options but when I plug in my Xbox one controller the computer doesn't read it.

FranticRain, 2015-01-31:
Is the controller lighting up? I did a poor job of programming the Preference Pane, so it won't recognize that a controller is plugged in unless its plugged in before the Preference Pane is opened. What really matters is if the controller is lighting up or not, because if it is, the controller is working.

Scorpian4875, 2015-01-31:
When I plug it into the computer it does not light up. The controller is less than a month old and barely used so there is definitely nothing wrong with it. Do I have to have to use some special type of micro usb chord or something?

FranticRain, 2015-01-31:
Check: Apple Menu -> About This Mac -> System Report -> Hardware -> USB. Do you have a device called "Controller" on that list?

Scorpian4875, 2015-01-31:
There's something called "USB Host Controller" but I'm not sure if that's it.

Scorpian4875, 2015-01-31:
Correction, "Bluetooth USB Host Controller"

FranticRain, 2015-01-31:
You are using a charge cable to connect the controller. Charge cables transfer no data, therefore the controller doesn't work. Try using a different cable to use the controller.

Scorpian4875, 2015-01-31:
Fortunately I have many a usb chord so I'll look around.

Scorpian4875, 2015-01-31:
I have found a cable that connects and the controller is under the system report as simply, "controller." I will follow previous instructions from there. Thanks!

FranticRain, 2015-01-31:
If the system sees the controller, and it lights up, then you should be in business. Just run any game with controller support.

Scorpian4875, 2015-01-31:
I'm trying to use it with skyrim (I wrapped it and able to play it on Mac) but the game doesn't recognize the controller. Does the game just not have controller support?

FranticRain, 2015-01-31:
Hmmm. I haven't tested the controller at all with Wine. Let me have a look at some stuff and then I'll get back to you.

Scorpian4875, 2015-01-31:
Alright thanks! Great support by the way!

Scorpian4875, 2015-01-31:
Nevermind I found a mod that makes it work. Thanks!

FranticRain, 2015-01-31:
Awesome! I'll go ahead and close this issue then. Have fun!

closed 2015-01-31 by FranticRain