Drewbadour/Xone-OSX #35: the Force Feedback plugin only seems to be found by OSX if the driver is in /System/Library/Extensions

geowar1, 2015-02-09:
Yeah, this is hard-coded in the sources; my work-around has been to put it in the right place (ok in driver package in /Library/Extensions/) and then symlink to it from /System/Library/Extensions/.

FranticRain, 2015-02-09:
See [this issue.](https://github.com/d235j/360Controller/issues/22)

RougeTomate, 2015-04-01:
@geowar1 - Could you please detail the work-around to make force feedback work? Thanks.

geowar1, 2015-04-01:
Install it in /Library/Extensions/ and then create a symlink to it in /System/Library/Extensions…
term> cd /System/Library/Extensions/
term> ln -sf /Library/Extensions/MyFFPlugIn.kext MyFFPlugIn.kext

RougeTomate, 2015-04-01:
Thanks. Not knowing how to do a symlink, I suppose I could use SymbolicLinker to perform the task.