Drewbadour/Xone-OSX #36: Need help uninstalling

hodsoni, 2015-02-10:
I noticed that there was a closed request for uninstalling already but I'm still having trouble. The README says there's a dmg inside the folder that uninstalls, but I don't see anything.

FranticRain, 2015-02-10:
[The latest release of the driver](https://github.com/FranticRain/Xone-OSX/releases/tag/v0.4.2) is a .dmg and includes an ```uninstall.command```. Simply double click on it, and it will uninstall the driver.

KingDuck, 2015-08-12:
What if you have an older version and want to uninstall it? Can you install the later version without uninstalling the older one?

FranticRain, 2015-08-14:
@KingDuck The uninstaller uninstalls every version of the driver. Additionally, every version of the driver can be installed over the previous version without problem.

closed 2015-08-14 by FranticRain