Drewbadour/Xone-OSX #37: Controls not working with Lego Marvel Superheroes Steam Version

ronaldooeee, 2015-02-13:
Xone-OSX works perfectly with Borderlands The Pre Sequel, Steam version. However, when I try using the controller for Lego Marvel Superheroes, the controller is recognized, but the button layout does not match the default controller settings in the main menu. For example, it says "Press A to accept", but when pressing "A", nothing happens. Instead, the "A" button is mapped out to the "X" button. Even after going to settings and resetting it to default, there is no change in the button mapping. This is only in the Main Menu. 

When actually starting the game, majority of the controls are completely different that what it's supposed to be. The start button on the controller makes my character move to the right. Left and Right on the D-pad & the triggers move the camera. Clicking the left stick ends up pausing the game. The "light button" on the controller makes my character punch. The face buttons are all mixed up, and that's pretty much it. It's unplayable with the controller.

FranticRain, 2015-02-13:
Uninstall my driver and [try using this driver](https://github.com/d235j/360Controller/issues/28#issuecomment-72572476). Let me know if that fixes it.

ronaldooeee, 2015-02-14:
Yes it worked! It even worked for Borderlands 2. However I do want to point out that initially, my character in Lego Marvel & Borderlands 2 would keep moving backwards & the camera could keep rotating down without me even touching the controller. Once I did move my character though, the moment I let go of the left analog stick, it would continue to move backwards. Same issue with the camera. The only fix that seemed to work for me was to unplug the controller during the game & then plugging it back in

FranticRain, 2015-02-14:
Yeah, its a known issue. Thankfully it works properly when I integrate into that driver. So as soon as they put out a release with my Xbox One controller support in it, I'll be pointing everyone that way. Glad it worked out for you. Have fun!

closed 2015-02-14 by FranticRain

ronaldooeee, 2015-02-14:
Thank you for your help & for your work! I appreciate it!