Drewbadour/Xone-OSX #38: HID element IDs collide and are inconstant with xbox 360 using Tattiebogle driver

alanu, 2015-02-17:
Since this driver reports the xbox one controller as an xbox 360 controller it should use the same hid ids for elements and same max and mins

for example : 
xbox one id 17 A button on xbox 360 17 is L3
xbox one id 18 B button on xbox 360 18 is L1
xbox one id 10 menu button on xbox 360 10 is dpad up
xbox one id 19 X button on xbox 360 id 19 is R1
xbox one id 20 Y button on xbox 360 iid 20 is the xbox button 

Also there max and min are different 
xbox one id 25 left  trigger return min 0 max 1023 on xbox 360 id 25 left trigger min 0 max 255
xbox one id 26 right trigger return min 0 max 1023 on xbox 360 id 26 right trigger min 0 max 255

This make supporting both controller in games very difficult since the both report as the same vendor and product ID and the above elements collide.

They should either match or report different product ids.

FranticRain, 2015-02-17:
If you are using Tattiebogle, you should just use [the Tattiebogle with Xbox One controller support.](https://github.com/d235j/360Controller/issues/28#issuecomment-72572476.)

The triggers on the Xbox One controller are much higher quality, and so I want that to be available to games. Other than that, the report descriptor pretty much copy pasted from Tattiebogle. I'm not sure where your button IDs are coming from. There are only 15 buttons total.

alanu, 2015-02-17:
I am getting the ids from IOHIDElementGetCookie

geowar1, 2015-02-17:
Yeah, the cookies aren't consistent across versions of OS X; use the usage page & usages; these are 16-bits so if you want a unique 32-bit number I'd just "(page << 16) || usage".

alanu, 2015-02-17:
Good to know thank for that info. looks like after changing id ever thing works thanks.

closed 2015-02-17 by alanu