Drewbadour/Xone-OSX #39: 3rd Party Controller support

reedptaylor, 2015-02-22:
Is there any way you can get this controller working? Its the PowerA Spectra Illuminated Controller for Xbox one.   Product ID:	0x542a
  Vendor ID:	0x24c6

FranticRain, 2015-04-13:
I'm so so so so sorry for missing this issue for so long. [This version of the driver](https://github.com/FranticRain/Xone-OSX/releases/tag/v1.0.1) should work with your controller. Let me know if you have any more problems.

reedptaylor, 2015-04-14:
Still no good:/ Controller just keeps blinking and the software doesn't see it

FranticRain, 2015-04-14:
Can you make sure that your controller shows up in the USB section of System Report when it is plugged in with or without the driver?

geowar1, 2015-04-14:
FranticRain: You should post an FAQ/troubleshooting guide; I'd guess about half the issues reports here are power-only USB cables. Most of the rest are "update to latest driver".

reedptaylor, 2015-04-14:
Yeah it has shown up there since before I installed the driver!

FranticRain, 2015-04-14:
So basically what this is telling me is that your cable is good and OS X recognizes the controller, but the driver isn't matching to it. This might be an outrageous question, but have you rebooted? As far as I understand, the controller should have matched without problem.

reedptaylor, 2015-04-14:
Rebooted and reinstalled

FranticRain, 2015-04-14:
Okay, let me compile a version with debug output enabled, and then we will see what the problem is.

FranticRain, 2015-04-14:
Okay, install this version, and then look at Console.app to see output from the driver. https://www.dropbox.com/s/sox8m8ktdvn4iik/Xone-Driver-360-1.0.1.dmg?dl=0

reedptaylor, 2015-04-14:
alright I'll let you know when I get home. Appreciate it!

reedptaylor, 2015-04-14:
Let me know if this is what you need: https://www.dropbox.com/sh/m3k2056ahfzshfu/AADLIvdbLo5ejsCwVL-fNPh2a?dl=0

FranticRain, 2015-04-14:
I need the "All Messages" section of Console.app located in Applications/Utilities. It looks roughly like this: https://www.dropbox.com/s/1nr4asmdngxcljw/Screenshot%202015-04-13%2023.04.37.png?dl=0

reedptaylor, 2015-04-14:
Alright I updated the dropbox folder

FranticRain, 2015-04-14:
Since I'm not even seeing anything from the driver in the log, the controller isn't even pairing with the driver.

There is some issue with PowerA controllers that my driver doesn't like. I'm not sure what it is.

@geowar1 Do you have any idea why the PowerA controllers would refuse to even get to ```start```?

FranticRain, 2015-05-29:
Please uninstall my driver and try using this beta version of this driver instead: https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview7.pkg

I've been porting my code into it, and it definitely works with the Microsoft controller. Hopefully it also works with third party controllers.

reedptaylor, 2015-05-29:
It works!!!!

FranticRain, 2015-05-29:
Awesome! Fantastic. When the official version of that driver comes out, this one will be deprecated. In the future, please use [the 360Controller repsitory](https://github.com/d235j/360Controller) for all of your Xbox One controller needs. I'll be around there if you have any further problems.

closed 2015-05-29 by FranticRain