Drewbadour/Xone-OSX #41: Bastion (Steam)

mattberkowitz, 2015-03-08:
For Bastion on Steam when using the Xbox One driver the controller is detected however buttons are all jumbled:

```
LB -> A
RB -> B
LS -> X
RS -> Y
Y -> LS
X -> Back
A -> D-Rt
Window -> D-Lt
Menu -> D-Dn
D-Lt -> RB
D-Up -> RS
D-Dn -> LB
RS-Left -> LT
RS-Up -> RT
Connect -> D-Up
```

FranticRain, 2015-03-09:
The 360 version of the driver is meant for game compatibility. I sent an email to Supergiant Games a couple months ago trying to see if they would add support for the traditional bindings of the Xbox One controller and they simply ignored me. This isn't so much an issue because the 360 version is doing its job perfectly, and the Xbox One version is acting as it should.

mattberkowitz, 2015-03-12:
Is it possible to have both drivers install simultaneously or I have to uninstall one before installing the other?

FranticRain, 2015-03-12:
You can not have both installed at a time, however I have no idea what will happen if you install both. Likely it will overwrite the previous installation.