Drewbadour/Xone-OSX #42: v1.0.0 not Working

sexto, 2015-03-17:
With the last Alpha, all games I tried worked perfectly, but with the latest release, even the controller appears in The Xone Controller Menu under System Preferences and works when I push all the buttons, none works. I tried Tomb Raider, VVVVVV and Don't Starve Together. All the Steam version.

FranticRain, 2015-03-17:
Are you using the Xone-Driver-1.0.0-rc.dmg or the Xone-Driver-360-1.0.0-rc.dmg install of the driver?

sexto, 2015-03-17:
Xone-Driver-1.0.0 (not rc).

Thanks for the fast feedback.

FranticRain, 2015-03-17:
Use the -360 version. The non-360 version is for people who want the raw mapping of the Xbox One controller. It will not work with most games, but people asked for it, so its there. The -360 version works with most games that support the Xbox 360 Controller.

sexto, 2015-03-17:
Yes. It works!

Thank you. Good job.

closed 2015-03-17 by FranticRain