Drewbadour/Xone-OSX #43: Prefrences Page blurred out

damagingbroom, 2015-03-24:
the preference option is blurred out for mac. I tried deleting this one file in the resources but it still couldn't find the controller.

FranticRain, 2015-03-24:
Does the controller light up when you plug it in? What did you delete? When the controller is plugged in, do you see a device called "Controller" in Apple Menu -> About This Mac -> System Report -> Hardware -> USB?

damagingbroom, 2015-03-24:
The controller does not light up when i plug it in, I can see it in the apple menu thing and it is a xbox one mini series controller

damagingbroom, 2015-03-24:
The thing i deleted was: NSPrefPaneGroups

FranticRain, 2015-03-24:
In that menu, what are the product and vendor id's listed for the controller?

damagingbroom, 2015-03-24:
Xbox ONE mini:

  Product ID:	0x541a
  Vendor ID:	0x24c6
  Version:	1.01
  Serial Number:	000074E3CE9A62BE
  Speed:	Up to 12 Mb/sec
  Manufacturer:	BDA
  Location ID:	0x14200000 / 7
  Current Available (mA):	500
  Current Required (mA):	Unknown (Device has not been configured)

FranticRain, 2015-03-24:
That controller is included in the info.plist, so it should work. Can you try connecting it to the computer with a different USB cable?

damagingbroom, 2015-03-24:
Do you mean a different port on my mac book pro or a different cable?

FranticRain, 2015-03-24:
Different cable

damagingbroom, 2015-03-24:
It has a built in wire so that isn't possible

FranticRain, 2015-03-24:
Could you run this command from your terminal and tell me what the output is? ```sudo kextutil "/Library/Extensions/Xone Driver.kext"```

damagingbroom, 2015-03-24:
it made me put in my password so I did and then nothing happened

FranticRain, 2015-03-24:
Are you running the latest version of the driver? It should give you a warning like: ```Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        XboxOneControllerDevice```

damagingbroom, 2015-03-24:
.9.1? because that is what i am running but remember the option is blurred out in system preferences until I delete the NSPrefPaneGroups file

FranticRain, 2015-03-24:
Not your version of OS X. Your version of he driver. https://github.com/FranticRain/Xone-OSX/releases/tag/v1.0.0

damagingbroom, 2015-03-24:
When i press refresh right next to where the controller selection box is it doesn't say anything and it is still blank so there is no warning

FranticRain, 2015-03-24:
Is the controller connected directly to the computer or to a hub? Try to connect it to computer if thats possible. Does the controller light up when you connect it?

damagingbroom, 2015-03-24:
What is a hub? and it is connected to the side usb (I have a retina mac book pro)

FranticRain, 2015-03-24:
Does the controller light up when you connect it?

damagingbroom, 2015-03-24:
No

FranticRain, 2015-03-24:
Then its not the driver's fault. You have a broken controller.

damagingbroom, 2015-03-24:
I don't think so because my friend got the same controller and it doesn't work either.

FranticRain, 2015-03-24:
Other users have used that specific controller with this driver from what I can tell. Do either of you have an xbox one that you have used the controller with? If its not lighting up, its either broken or not getting power. OS X automatically powers controllers. It is out of my control.

damagingbroom, 2015-03-24:
I haven't tried it on my xbox one yet... there is a part on the cord that I can disconnect and is a circle shape

FranticRain, 2015-03-24:
Let me know if it works with an Xbox One. There is nothing more I can do unless I know if it works or not.

damagingbroom, 2015-03-30:
It does work for xbox one btw

FranticRain, 2015-03-30:
https://github.com/FranticRain/Xone-OSX/releases/download/v1.0.0-rc/Xone-Driver-For-Debugging.kext.zip
Unzip this, place it in ```/Library/Extensions``` then run ```sudo kextutil "/Library/Extensions/Xone Driver.kext"``` then look at the Console application found in the utilities folder of applications. Is there anything printed that contains "Xbox One Controller"?

damagingbroom, 2015-03-30:
There is no extensions page

FranticRain, 2015-03-30:
I have no idea what you mean.

Every version of OS X from 10.9 to later has a /Library/Extensions folder.

RougeTomate, 2015-04-01:
@damagingbroom - search the library folder located at: Macintosh HD/Library, not the one under your username.

FranticRain, 2015-05-29:
Please uninstall my driver and try using this beta version of this driver instead: https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview7.pkg

I've been porting my code into it, and it definitely works with the Microsoft controller. Hopefully it also works with third party controllers.