Drewbadour/Xone-OSX #48: I have a question

TheModMaster8, 2015-04-19:
Can you use Xone-OSX Driver in eclipse IDE  for example say i have a robot and want to make it go forward, can i assign it to the joystick in eclipse ?

FranticRain, 2015-04-19:
I'm unfamiliar with Eclipse's joystick integration. Where in Eclipse can I find this? Did you download a plugin to Eclipse that allows you to use a controller? Or are you writing code that interacts with a controller?

TheModMaster8, 2015-04-19:
I'm just trying to interact with the data values given off by the joystick, i want to take those outputs and make eclipse read them and give an output of some kind, i just don't know how to call the values being given off by the controller,  also I'm unaware of any joystick integration within eclipse. or of any plugins that would aid in my goal.

FranticRain, 2015-04-19:
Eclipse isn't going to get controller input for you. You need to write a program that gets controller input. It depends on what programming language you are using with Eclipse though. If you are using Java, you might consider a library such as [LWJGL](http://www.lwjgl.org/). And do implement it, you might [find this useful.](https://youtu.be/jwFzrBiH8X0)

TheModMaster8, 2015-04-19:
Thanks for the help  ill see if this will work, it look promising!

FranticRain, 2015-04-19:
You will need a driver before LWJGL will see the controller though, just remember that. Same thing applies to Xbox 360 Controllers. The bindings for LWJGL likely look for 360 Controller style button layouts, so it would probably be a good idea to use the Xone-Driver-360-X.X.X version of the driver.

closed 2015-05-06 by FranticRain