Drewbadour/Xone-OSX #49: System preferences crashes when I hit Xone icon

nicktothek, 2015-04-30:
not only that the controller will not turn on when connected, it shows up in the system report and all but will not work!! please help

FranticRain, 2015-04-30:
Please follow the troubleshooting guidelines and then let me know what happens: https://github.com/FranticRain/Xone-OSX#troubleshooting

nicktothek, 2015-04-30:
It shows up in system report, but it doesn't light up when I plug it in. Also like I said in the title I click the Xone icon on the system preference page and it crashes and says "system preferences has quit unexpectingly"

FranticRain, 2015-04-30:
Are your product and vendor IDs the same as the example?

nicktothek, 2015-04-30:
Yeah

FranticRain, 2015-04-30:
What version of OSX and what version of the driver are you using?

nicktothek, 2015-04-30:
Im running 10.8.5 and I'm using Xone 4.0

FranticRain, 2015-04-30:
This driver is not officially supported on version of OS X earlier than 10.9. If you want to try and run it on 10.8.5, copy "Xone Driver.kext" from /Library/Extensions to /System/Library/Extensions.

closed 2015-04-30 by FranticRain

nicktothek, 2015-04-30:
what driver is supported on 10.8.5?

FranticRain, 2015-04-30:
This one may: https://github.com/guilhermearaujo/xboxonecontrollerenabler

nicktothek, 2015-04-30:
I can only find the version supported by 10.9

FranticRain, 2015-04-30:
Since OS X driver stuff changed with 10.9, and the Xbox One controller was released after 10.9. So no one wants to support it.