Drewbadour/Xone-OSX #51: product and vendor ID

trdavi2, 2015-05-06:
Product ID:	0x542a
  Vendor ID:	0x24c6

FranticRain, 2015-05-06:
What is the name of this device, for my records.

FranticRain, 2015-05-06:
Oh. Its a PowerA Spectra Illuminated controller. Do you know you to use a USB packet sniffer in windows?

trdavi2, 2015-05-06:
No and I'm on a mac.

----------------------------------------------------------------
This message was sent using Illinois State University RedbirdMail

FranticRain, 2015-05-06:
That controller is already supported in code. However, it does not follow the standards for starting an Xbox One controller. It does not work with this driver until someone with a Windows machine can test it with a packet sniffer.

FranticRain, 2015-05-29:
Please uninstall my driver and try using this beta version of this driver instead: https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview7.pkg

I've been porting my code into it, and it definitely works with the Microsoft controller. Hopefully it also works with third party controllers.

closed 2015-10-05 by FranticRain