Drewbadour/Xone-OSX #52: Triggers

Wevz, 2015-05-09:
I'm having issues when it comes to the mapping of the triggers.

It seems when i am assigning them, i press them and its almost as if they get stuck. When using them in Steam's big picture they then become unresponsive or whenever i click to assign another button the trigger's axis takes its place as if i am still pressing it. I have tried it on Verdun on steam and also Westerado:Double barrelled.

Apart from that i'm so happy i've found something to use a controller, thank you!!

FranticRain, 2015-05-09:
I'm not quite sure off the top of my head what the problem is. Are you using a third party controller or the official Microsoft one? Make sure you are using the latest version of the driver "Xone-Driver-360-1.0.1.dmg"

Wevz, 2015-05-09:
Will give that a go cheers!