Drewbadour/Xone-OSX #53: Please add this product and vendor ID

ciaskohansen, 2015-05-09:
ID del producto:	0x0063
  ID del fabricante:	0x0f0d  (HORI CO., LTD.)

REAL ARCADE PRO. ONE

Thanks

FranticRain, 2015-05-09:
[Let me know if the latest release of the driver works for you.](https://github.com/FranticRain/Xone-OSX/releases/tag/v1.0.2)

ciaskohansen, 2015-05-09:
Do not work

FranticRain, 2015-05-10:
Could you please post the contents of Console.app after you plug in the controller?

ciaskohansen, 2015-05-10:
I have used driver 1.0.2
I add the log

-- 
Cisko
Sent with Airmail

ciaskohansen, 2015-05-10:
I hope your help
thanks

FranticRain, 2015-05-11:
I'm taking a couple of shots in the dark here. Could you test each one of them?
Test 1: https://dl.dropboxusercontent.com/u/33409904/Xone-Driver-1.0.2-r1.dmg
Test 2: https://dl.dropboxusercontent.com/u/33409904/Xone-Driver-1.0.2-r2.dmg

ciaskohansen, 2015-05-11:
not work, I have tried r1 and r2 .The led of the real arcade stick do not turn on .
I add the log
[image]

ciaskohansen, 2015-05-11:
now the led of the stick is on, but the same error in console.app

and it is not appear int he menu

[image]

FranticRain, 2015-05-11:
Which version turned the LEDs on? And did you try pressing the refresh button to get the controller to appear in the menu? Also try the controller with something like Steam Big Picture. The failure to appear in the system preferences may have just been sloppy programming on my part.

ciaskohansen, 2015-05-11:
now the led do not turn on, it is rare, I connect the cable and disconnect
and the same problem.
I refresh but nothing, do not appear.

ciaskohansen, 2015-05-13:
something new?

FranticRain, 2015-05-13:
Not yet. When the LEDs turn on, does it stay solid? Or is it just blinking? Because if it starts to go solid, there might be some useful data in Console.app.

ciaskohansen, 2015-05-13:
the xbox logo do not turn,but  the stick has a small led  when it is
working through usb .

ciaskohansen, 2015-05-13:
I tried r4 and nothing

Led turns on.

[image]

[image]

[image]

FranticRain, 2015-05-13:
r4 doesn't change anything on your end, so it wouldn't fix anything. If there was data in the console, it would be before the "Failed To Connect" lines, but the issue appears to be with pairing. I'll try and get something to you soon that will help me debug the problem.

FranticRain, 2015-05-14:
Okay, try this out and let me know what score you get in Console.app.
https://dl.dropboxusercontent.com/u/33409904/Xone-Driver-1.0.3-r1.dmg
The format should be "Score - XXXXX". For example, mine is "Score - 90000"

ciaskohansen, 2015-05-19:
ok, I am going to try

FranticRain, 2015-05-19:
Sorry about that. Should be fixed now.

ciaskohansen, 2015-05-19:
where is about "score"?

FranticRain, 2015-05-19:
It is in your Console.app output.
[image]

ciaskohansen, 2015-05-19:
but I do not see it

FranticRain, 2015-05-19:
You installed the driver, and plugged in the controller and then checked Console.app and there was not output listing the score? Could you please run these commands in the terminal:
```cd /Library/Extensions```
```sudo kextuil "Xone Driver.kext"```
And then plug the controller in and check again.

ciaskohansen, 2015-05-19:
If I connect xbox one controller is all fine, the score appears

If I connect the stick nothing

-- 
Cisko
Sent with Airmail

ciaskohansen, 2015-05-19:
"sudo: kextuil: command not found"

FranticRain, 2015-05-19:
I made a typo, it should be "kextutil" but thats not important, I found the issue that was preventing you from getting a core. Try this version: https://dl.dropboxusercontent.com/u/33409904/Xone-Driver-1.0.3-r2.dmg

ciaskohansen, 2015-05-19:
not working
19/5/15 23:41:11,349	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:11,355	com.apple.xpc.launchd[1]	(com.apple.StreamingUnzipService) The _DirtyJetsamPriority key is not available on this platform.
19/5/15 23:41:11,355	com.apple.xpc.launchd[1]	(com.apple.StreamingUnzipService) The _DirtyJetsamMemoryLimit key is not available on this platform.
19/5/15 23:41:11,359	storedownloadd[424]	DownloadServiceDelegate: Accepting new connection <NSXPCConnection: 0x7fd8c2d03c20> connection from pid 413 with interface <DownloadServiceInterface: 0x7fd8c2d061d0> (PID 413)
19/5/15 23:41:11,551	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:11,734	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:11,861	storedownloadd[424]	DownloadServiceDelegate: Accepting new connection <NSXPCConnection: 0x7fd8c2e00460> connection from pid 413 with interface <DownloadServiceInterface: 0x7fd8c2e008c0> (PID 413)
19/5/15 23:41:11,885	storeassetd[413]	AssetServiceDelegate: Accepting new connection <NSXPCConnection: 0x7fb984211370> connection from pid 424 with interface <AssetServiceInterface: 0x7fb984216420> (PID 424)
19/5/15 23:41:11,951	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:12,187	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:12,436	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:12,870	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:13,088	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:13,405	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:13,639	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:13,873	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:14,475	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:17,032	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:17,232	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:17,359	com.apple.SecurityServer[76]	Session 100004 created
19/5/15 23:41:17,433	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:17,633	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:17,817	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:18,018	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:18,218	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:18,385	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:18,569	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:18,752	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:18,937	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:19,137	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:19,304	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:19,505	System Preferences[376]	JOYSTICK COUNT - 0
19/5/15 23:41:21,064	SpotlightNetHelper[336]	[SLSUGGESTIONS] Location managed failed with error Error Domain=kCLErrorDomain Code=0 "No se ha podido completar la operación. (Error de kCLErrorDomain 0.)"
19/5/15 23:41:24,351	SpotlightNetHelper[336]	336: CFNetwork internal error (0xc01a:/SourceCache/CFNetwork/CFNetwork-720.3.13/Connection/URLConnectionLoader.cpp:1321)
19/5/15 23:41:24,352	loginwindow[88]	Preferred Localizations total: 1 contents (
    es

ciaskohansen, 2015-05-19:
and it is rare but the led of the stick only turns on if I connect it before turning on the Mac

ciaskohansen, 2015-05-19:
Notice: /Library/Extensions/Xone Driver.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        MicrosoftDevice
        PDPDevice
        HoriDevice
        PowerADevice

iMac-de-Cisko:Extensions Cisko$

FranticRain, 2015-05-19:
Okay, one last version of the driver. Let me know if it helps: https://dl.dropboxusercontent.com/u/33409904/Xone-Driver-1.0.3-r3.dmg

ciaskohansen, 2015-05-21:
not working.........what is  the problem?

[image]

FranticRain, 2015-05-29:
Please uninstall my driver and try using this beta version of this driver instead: https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview7.pkg

I've been porting my code into it, and it definitely works with the Microsoft controller. Hopefully it also works with third party controllers.

closed 2015-10-05 by FranticRain