Drewbadour/Xone-OSX #54: Xbox One mini?

patathom, 2015-05-11:
I bought a Xbox One mini controller. It's on the list but it isn't detected in the program. Any solution?

P.S I'm running it on Yosemite 10.10.3.Thank you
Thank you

FranticRain, 2015-05-11:
I haven't gotten 3rd party controller support down quite just yet. So try this experimental version of the driver and let me know if it works.

https://dl.dropboxusercontent.com/u/33409904/Xone-Driver-1.0.2-r3.dmg

Also make sure to follow the troubleshooting instructions and make sure that your controller is showing up as connected, and you don't have a bad cable.

patathom, 2015-05-12:
First, thank you very much for answering this quick.

The other program is suppose to change the previous in the settings or to install a new one?
Because it isn't working.
For the troubleshooting, my controller is in the menu, but it is not detected in the program.

FranticRain, 2015-05-12:
The installer should have overwritten the first driver with a new one. What you are really looking to see is whether or not your controller lights up. Does your controller light up at all?

patathom, 2015-05-12:
Nope, not at all

patathom, 2015-05-12:
do I need to uninstall the first?

FranticRain, 2015-05-12:
No. The new installer will overwrite the first one. Try this other one. Maybe it will yield different results. https://dl.dropboxusercontent.com/u/33409904/Xone-Driver-1.0.2-r4.dmg

If It doesn't work, Look in Console.app located in Applications/Utilities. Let me know if you see anything in there that looks like: "Xbox One Controller - handleStart"

patathom, 2015-05-12:
When i finished installing the program, it's written that it cannot localize the necessary data. In the consol, there's a lot of "unmount initiated on xBox One Driver of /Volume/Xbox One Driver for OS X 1". My cable is good, it work on my Xbox One.

FranticRain, 2015-05-12:
You need to look at the console directly after plugging the controller in. I need to know what's happening when the controller is plugged in.

patathom, 2015-05-12:
"Spindump: invalid connection: com.apple.coresymbolicationd" and "com.apple.xpc.lunchd"

"The_DirtyJetsamMermoryLimit key is not available on this platform"

FranticRain, 2015-05-12:
The driver isn't even seeing the controller. Can you just make sure that your Vendor ID is 0x24C6 and your Product ID is 0x541A ?

patathom, 2015-05-14:
Yes it's that

FranticRain, 2015-05-14:
Okay, try this out and let me know what score you get in Console.app.
https://dl.dropboxusercontent.com/u/33409904/Xone-Driver-1.0.3-r1.dmg
The format should be "Score - XXXXX". For example, mine is "Score - 90000"

Badbubba, 2015-05-22:
Hi, i am experiencing same problem as patathom describes. Using Xbox one mini Vendor ID 0x24c6 and product id is 0x541A, Mac running Yosemite 10.10.03. Any progress on solving this issue ?

FranticRain, 2015-05-22:
There is no new progress. I'm not sure what the problem is, I'm working on something that might fix it though.

FranticRain, 2015-05-29:
Please uninstall my driver and try using this beta version of this driver instead: https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview7.pkg

I've been porting my code into it, and it definitely works with the Microsoft controller. Hopefully it also works with third party controllers.

patathom, 2015-05-31:
I couldn't find the "uninstall.command" if I re-download the original program, will it be there?

FranticRain, 2015-05-31:
Yes. The uninstaller is included in all of the latest versions and will uninstall any version of the driver.

patathom, 2015-06-01:
It worked thank you so much for all the help!!

FranticRain, 2015-06-01:
Awesome! Fantastic. When the official version of that driver comes out, this one will be deprecated. In the future, please use the 360Controller repsitory for all of your Xbox One controller needs. I'll be around there if you have any further problems.

closed 2015-06-01 by FranticRain