Drewbadour/Xone-OSX #55: Uninstall

namzat18, 2015-05-13:
Now i know you have said that there is a uninstall.command in the .dmg but where?????????? I really don't see it and I'm starting to get a little mad because it really just doesn't want to work so I'm done. now tell us a step by step of how to uninstall it because saying its there and it not being there is bs.

FranticRain, 2015-05-13:
1. Download the latest version of the driver. Actually, pretty much any version will do.
2. Double click on the downloaded .DMG. Ex. Xone-Driver-1.0.2.dmg. This file should be located in your downloads folder.
3. DMG will open.
4. Click on the uninstall.command file. It should be in the very top left of the newly opened window.
This is what the opened DMG should look like:
[image]

namzat18, 2015-05-13:
Thank you very much sir.

closed 2015-05-13 by namzat18