Drewbadour/Xone-OSX #56: Please add southpaw support

joshuatruscott, 2015-05-13:
Hello,
First off, let me say that I have been using Xone-OSX for a week now and love it. Thank you.
Just one request, could you add southpaw (swap the sticks) support? I have cerebral palsy and playing shooters without southpaw is very difficult for me. Also a lot of lefties will thank you ;)

FranticRain, 2015-05-13:
I really can't believe that I didn't think of doing this before. Its a great idea. I started hacking on it this morning, and it should have been really easy to implement. But let me tell you, Preference Panes are the devil and I want to try the person who created them for crimes against humanity!

geowar1, 2015-05-13:
What's hard about Preference Panes? With a NSUserDefaultsController and bindings you can do almost everything without any code at all. Although looking at yours you do get all fancy with the artwork… (and you call yourself an engineer! ;-).

FranticRain, 2015-05-13:
Well the problem is, I make some changes, and compile, open it up and it works! Everything is great! Okay lets add a line that declares a constant NSString. And compile again, and open it up and it doesn't load. I quit out of System Preferences and it crashes. Okay, so lets remove the constant NSString, maybe that did it! Okay now it doesn't work at all again. Okay so let's just trash all of the changes, and try it again. Well that didn't work either. Okay lets download the code from the repo and try just that raw code that I know works. Oh wait that doesn't work either! There is absolutely no rhyme or reason, and as far as I can tell, no way to debug the insanity.

And for the record, a friend of mine did the art, not me. :P

FranticRain, 2015-05-13:
And again. I ran a build. Everything worked great! Then I go in to change the Preference Pane icon label. The label changes, but now the content of the Pref Pane doesn't work. So I undo that, compile again, and  now the label is back to being wrong, and the content still doesn't work. I can't possibly comprehend what is wrong. Its just a complete mess that is not suitable for anyone to use.

geowar1, 2015-05-13:
> On May 13, 2015, at 9:31 AM, Drew Mills <notifications@github.com> wrote:
> And again. I ran a build. Everything worked great! Then I go in to change the Preference Pane icon label. The label changes, but now the content of the Pref Pane doesn't work. So I undo that, compile again, and now the label is back to being wrong, and the content still doesn't work. I can't possibly comprehend what is wrong. Its just a complete mess that is not suitable for anyone to use.
> —

I took a look but didn’t see anything obvious… that didn’t stop me from re-writing it thou… 

You’ve got a lot of code to convert between NSStrings and CFStringRef’s… that’s unnecessary because you can just bridge cast them (they’re toll-free bridged). 

Same thing for NSDictionary & CFDictionary (you can use Modern ObjC syntax to create a NSDictionary and then __bridge cast it to CFDictionaryRef! ;-)

Lots of getters and setters being called (as methods); changed those to direct access (faster and more readable). (Except on array elements where the type is “id”.)

Here’s the whole project… although all my changes are in the one source file.

BTW: How are you installing the Preference Pane after it’s (re-)built? Don’t Finder drag-copy to (~)/Library/PreferencePanes/. Instead double-click to open it; System Preferences should launch as ask if you want it installed for this or all users, etc.

That’s the safest way to install it because the System Preferences app “does the right thing” (™, inc. ;-).

-- 
Enjoy,
George Warner

geowar1, 2015-05-13:
If you can't get the enclosure email me at: <geowar 1 @ mac . com> (no spaces) and I'll send it to you directly.

FranticRain, 2015-05-13:
Here you go! 1.0.3 should have added Southpaw support. Let me know if it works for you. https://github.com/FranticRain/Xone-OSX/releases/tag/v1.0.3

joshuatruscott, 2015-05-13:
Thank you so much! It works great!

closed 2015-05-13 by FranticRain