Drewbadour/Xone-OSX #57: Javascript API issues

Chrisui, 2015-05-16:
Hey,

I've installed the latest driver and my mac seems to register that a gamepad is connected just fine.

I'm developing a javascript game engine and am having some trouble using an xbox controller on a mac (usually what I use to program) via this driver. Essentially when trying to access via the [Gamepad API](https://developer.mozilla.org/en-US/docs/Web/API/Gamepad_API) the controller values are never updated on a mac. Another issue seems to be that *2* controllers appear in the controller array (`navigator.getGamepads()`)

You can see the issue at this demo here: http://chrisui.github.io/titan-2d/examples/index.html

On a windows machine with the standard controller drivers you can move the ball around with the left analog stick and jump with 'A' - you need to press any button on the pad to register it first of all. On mac this doesn't happen as no new controller values are ever seen. You can see the values by opening up the browser console log (logged every 20th frame)

If you need any more information on reproducing this let me know!

Chris

FranticRain, 2015-05-16:
Have you tested this with a 360 Controller and the (360Controller driver)[https://github.com/d235j/360Controller]? I would test it myself, but I don't have a 360 controller around right now.

Chrisui, 2015-05-16:
Haven't got a 360 controller I'm afraid.

FranticRain, 2015-05-16:
I wouldn't even know what to touch without a check on the 360 driver first. Chrome has built in drivers for the Xbox One controller. I know that for sure. It might be a good idea to check it without my driver installed first. More info about that chrome driver is here: http://blog.tojicode.com/2014/02/xbox-one-controller-in-chrome-on-osx.html

Chrisui, 2015-05-16:
Interesting. Uninstalling this driver does allow the controller to work in Chrome!

Issue can be closed for my needs but feel free to keep it around incase you want to look into why the driver prevents the browser code from working.

closed 2015-05-16 by Chrisui