Drewbadour/Xone-OSX #58: Hero Siege - Buttons work incorrectly

Kimkjaer, 2015-05-18:
Hello,
First thanks for the time invested in this project ! :)

I downloaded this Program to play Hero Siege (Steam) with my xone controller..
But the buttons isn't responding like it should ..

The game is made with a option to use a controller - and has the buttons shown on screen but when i press them it doesn't react like it should ..

can you fix this ?? :)
Thanks again

FranticRain, 2015-05-18:
Hero Siege is a Game Maker game. My driver doesn't work with Game Maker games. But I've been working to add Xbox One controller support to a driver that does. So uninstall my driver, and then install this one, and your controller should work perfectly: https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview6.pkg

Let me know if it works.

Kimkjaer, 2015-05-18:
Worked perfectly with my xbox one controller even tho it said 360 Driver !! :-D

Thanks so much :+1:  
Do you take donations?
Because you just saved me the money from buying a ps4 controller :+1:

FranticRain, 2015-05-18:
No, I don't take donations. Mostly because this driver will be dead when they finally update 360Controller to officially support Xbox One controllers. I'd say hold onto that money and look into buying a cool wireless dongle for your controller when those come out. Have a great day.

closed 2015-05-18 by FranticRain

Kimkjaer, 2015-05-18:
All right - what kind of 'dongle' are you referring too ?
I use the original wireless xone controller - but ain't i supposed to use the wire to make it work ??

Or am i forced to buy an PS4 controller :-1:  :( ?

FranticRain, 2015-05-18:
The Xbox One controller only works wired to computers now. MS will eventually sell a dongle that will allow it to work wirelessly. Thats the way it is with Xbox One controllers.