Drewbadour/Xone-OSX #61: Incorrect mappings on README

jamierocks, 2015-06-04:
It seems the mapping are wronf on the README :|

I found that your Y mapping maps to UP and MENU maps to VIEW

My settings are default

FranticRain, 2015-06-04:
You're right and wrong. The bindings are not the ones you have. Those bindings are for the "OLDBIND" version of the driver, previously the default version of the driver. The bindings you have are identical to the Xbox 360 Controller bindings. Listing those bindings on the README seemed useless. So I will go back and fix the issues with the README not being clear about that. Do you recommend any specific changes to the README?

jamierocks, 2015-06-04:
Hmm, no I check I got the normal drivers...

jamierocks, 2015-06-04:
To be specific I downloadedhttps://s3.amazonaws.com/github-cloud/releases/24651348/48abc522-f999-11e4-9d15-d2655f1f8389.dmg?response-content-disposition=attachment%3B%20filename%3DXone-Driver-1.0.3.dmg&response-content-type=application/octet-stream&AWSAccessKeyId=AKIAISTNZFOVBIJMK3TQ&Expires=1432980062&Signature=UfQLzI9unRwTtpeOxJvGhAjpjS8%3D

FranticRain, 2015-06-04:
The default version of the driver is bound to be identical with Xbox 360 Controllers. It advertises itself as a 360 Controller and uses 360 Controller bindings to achieve the best possible compatibility with games.

jamierocks, 2015-06-04:
Ah, that makes sense. It would be great if you put the mappings some where :D

FranticRain, 2015-06-04:
Also note, if you are comparing these to Steam's Big Picture, they zero index their buttons, while USB's official documentation (and my documentation) one index the buttons. That would explain why your buttons would be off by one like you say.

jamierocks, 2015-06-04:
I'm using a controller library (provided by LWJGL) for my controller support :P

FranticRain, 2015-06-04:
If you are looking for longterm Xbox One controller support, I highly recommend that you focus on the [360Controller repository.](https://github.com/d235j/360Controller) I have been porting my code into that driver, because it works with third party controllers, while mine does not. Once they release an official version of that driver, mine will be completely deprecated. You can access the beta version of that driver from my releases page. The bold text talking about third party controllers should get you the right download. Don't forget to uninstall my driver when you install that one.

jamierocks, 2015-06-04:
Ok, having a look now. Thanks :)

Is there a list of HID mappings?

FranticRain, 2015-06-04:
If you're looking for all HID mappings, you'll need to look at the USB documentation. If you want to see my implementation its in the 360hid.h. I've been looking at the LWJGL implementation, and it looks like the buttons are zero indexed there as well. So you probably just need to take my table and subtract one from all of the numbers.

jamierocks, 2015-06-04:
Awesome, thanks :D

closed 2015-10-05 by FranticRain