Drewbadour/Xone-OSX #63: New Xbox One "Covert Forces" Controller

SmirksAlot, 2015-06-20:
Product ID:	0x02dd
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	2.03

This is the new controller with the 3.5mm headset jack build-in.

Could you please make sure this is added to the compatibility list of the driver as it appears to not work for me.

Thanks!

FranticRain, 2015-06-20:
Awesome! I was hoping someone would get one of these. I updated the driver so that it should support the controller. Check the releases page and let me know if it works.

SmirksAlot, 2015-06-20:
I downloaded 1.0.4.dmg. But inside is still 1.0.3? Is that correct?

FranticRain, 2015-06-20:
That will work fine. Its just a sloppy copy paste job by me, I'll fix it real quick.

FranticRain, 2015-06-20:
Actually, download the latest version again. Should be fixed. Its been so long since I've built, I forgot a couple steps!

SmirksAlot, 2015-06-20:
Thanks. I did. Controller one is now showing up. But I'm trying to
troubleshoot why none of the buttons are registering.

SmirksAlot, 2015-06-20:
When I plug it in. I get 4 kernel messages.

Kernel: USB (XHCI Root Hub USB 2.0 Simulation): Port 2 on bus 0xa connected
or dudcinnected: portSC(0xe0206e1)
Kernel: Score - 9000
Kernel: Xbox One Controller - handleStart
Kernel: Sandbox: com.apple.webkit(384) deny iokit-open IOHIDLibUserClient

Any ideas?

FranticRain, 2015-06-21:
Did you try it in any games? The preference pane can be finicky. Basically the top three lines of the console tell me that the controller start just fine. I have no idea what that fourth line is, but it seems related to webkit, which shouldn't be a problem.

SmirksAlot, 2015-06-21:
no light on the controller.

the mac knows it's an xbox controller (using joystick pro) but none of the buttons register when pressed

FranticRain, 2015-06-21:
It vibrates in OS X? I know it vibrates when you plug it in on windows, but it shouldn't vibrate in OS X.

SmirksAlot, 2015-06-21:
that's interesting... i have parallels installed on this mac, but it's not running...

FranticRain, 2015-06-21:
My driver would probably override Parallels. If you uninstall my driver, it should work with Parallels. You do need to download the windows driver manually if you aren't running Windows 8, however.

SmirksAlot, 2015-06-21:
hmm, but i'm running steam games on mac, not in parallels. Parallels is just for desktop apps, not my games.

am i stuck? parallels is messing with your driver?

FranticRain, 2015-06-21:
I can't say for sure. There are too many variables here. My driver should be working just fine, but it may have some interaction with Parallels. I don't know of anyone else who is using this driver and Parallels with the old controller, so I don't know.

SmirksAlot, 2015-06-21:
thanks anyway for all the help, i appreciate your time and effort!

FranticRain, 2015-06-21:
I just installed a trial version of Parallels and the old controller works just fine with the driver. So what I'm guessing is that Microsoft changed the way the controller works a little bit with this revision. The trick is going to be figuring out how. Unfortunately, I would need to sniff the usb packet on a windows system to figure it out. And since I don't have a controller, I can't do it myself. I don't recommend you do it either, because there aren't really any good free USB sniffers, and most of them have awful pack ins that will mess up a windows installation. There is one thing you can test though. [Go to this website](http://www.html5rocks.com/en/tutorials/doodles/gamepad/gamepad-tester/tester.html) and tell me if your controller lights up.

SmirksAlot, 2015-06-21:
Ok, I just installed your driver on my wife's macbook (which doesn't have parallels) and it works exactly the same as mine. It rumbled for a second when I plugged it in and it doesn't light up.

I then went to the website in chrome, and it didn't appear to work. I got the message...

"No gamepads seem to be connected. Be sure to plug in a gamepad and then press any of its buttons to activate it."

but in your driver it says controller 1 in the drop down list.

FranticRain, 2015-06-21:
So what happens is when you plug in a controller, the driver sends the controller some data to tell it to "wake up." This data also turns on the LED in the controller. In the old version of the driver, this equates to the data 5 and then 32. Unfortunately, it appears that this set of data isn't the same for this new controller. If it was, that website would have woken up the controller, no problem. Of course, if the driver is installed when you attempt to use that website, it prevents the website from doing its job.

The situation with the gamepad being listed is that the drive believes that it has sent the wakeup packet, but the controller did not wake up. So the driver is telling the operating system that a controller is connected, but the controller isn't sending any data.

So make sure to test that website WITHOUT the driver installed, and let me know if it works or if it doesn't.

SmirksAlot, 2015-06-21:
I uninstalled your driver, and went to the website in chrome, i haven't seen the website work yet. i tried with Xbox One, PS3 and PS4 controllers, nothing worked.

FranticRain, 2015-06-21:
How did you connect these controllers? The Xbox One controller and PS4 controller will convert to a wired controller when plugged in via USB. I got the website to work with both an Xbox One controller and a PS4 controller.

SmirksAlot, 2015-06-21:
I just plugged them in with a USB cable, PS4 controller works in games no problems, but doesn't work in the website. i'm confused now. i have the latest version of chrome.

FranticRain, 2015-06-21:
This is very curious. Try plugging the cable into the controller, opening the website, plugging the cable into the computer and then pressing the Cross (X) button on the PS4 controller.

SmirksAlot, 2015-06-21:
I just did, the ps4 controller glows, but still nothing on the website. Do i have to enable something in chrome to turn this on? permissions perhaps?

SmirksAlot, 2015-06-21:
OK, i deleted Joystick mapper and the website now works with PS4, XBox One still does nothing

FranticRain, 2015-06-21:
Okay, the packet has changed. Someone will have to sniff the new start packet before I can do anything. So there is nothing I can do.

SmirksAlot, 2015-06-21:
well, at least we know where the problem is! thanks again for working with me trying to figure this out.

FranticRain, 2015-06-21:
If you get impatient and want to sniff the packet for yourself, let Parallels take control of the Xbox One controller and then follow installation instructions in "Following along," and then tell me what the URB_INTERRUPT packets are like in the "The stuff magic is made of" section. You might also want to fiddle around with some buttons and see if the packets still start with "20."

KingSlayer816, 2015-06-22:
Hi FranticRain,

First off, thank you so much for this driver, it works very well and I appreciate your effort in making it. As SmirksAlot said above, my controller (the new covert forces one, not the black old one) only vibrates when plugged in and the X does not glow up. I don't have a deep understanding of packets (or any, for that matter), and i tried that website you mentioned, and my covert forces controller doesn't register. Have you figured out what the problem is with the new controllers?

Thanks for the help!

FranticRain, 2015-06-22:
Did you install the windows Xbox One Controller driver onto your windows
machine?

KingSlayer816, 2015-06-22:
I have a mac, osx yosemite

KingSlayer816, 2015-06-22:
I did download the driver, and it works perfectly with my old xbox one controller, which has an issue with the joystick, but the driver doesn't activate the covert forces controller.

FranticRain, 2015-06-22:
The packet sniffing technique only works on Windows.

KingSlayer816, 2015-06-22:
Oh ok, thanks, I'll probably have to wait until the new packets have been implemented into the driver to use the covert forces controller then?

FranticRain, 2015-06-22:
Correct. Hopefully someone with a windows machine will check it for me.

KingSlayer816, 2015-06-22:
Alright, Thanks!

StillScreaming, 2015-06-24:
The controller also doesn't work on Windows (7). Seems like the old driver doesn't work with the new controller and Microsoft need to put out a new one. When I plug it in it to Windows/OSX it just vibrates once and that's it.

FranticRain, 2015-06-24:
Well that's at least good news for me. My guess is that it won't be officially supported in anything other than Windows 10. That of course makes sniffing the packets harder until Win10 releases. So it just looks like this controller is going to go unsupported until then.

StillScreaming, 2015-06-24:
Yep it works fine in Windows 10 Preview

FranticRain, 2015-06-24:
Could you sniff the packet in Windows 10 Preview for me? [I mentioned how slightly further up in the thread.](https://github.com/FranticRain/Xone-OSX/issues/63#issuecomment-113965052)

Ideally, I need to know the first packets sent to the controller, what pressing buttons and moving sticks looks like, and maybe if you've got a game that will rumble the gamepad, those packets would be great.

StillScreaming, 2015-06-25:
I can't figure out the packets when connecting so here is link to a file where I connected then disconnected 3 times https://drive.google.com/file/d/0B6ICCfrPaX4mckFGVzYwMVVtTFk/view?usp=sharing
And here is what it looks like when pressing the buttons
Logo pressed 0720xx02015b released 0720xx02005b
a 2000xx0e100000000000f70732f893043d01 
b 2000xx0e200000000000f70732f893043d01
x 2000xx0e400000000000f70732f893043d01
y 2000xx0e800000000000f70732f893043d01
back 2000xx0e080000000000f70732f893043d01
start 2000xx0e040000000000f70732f893043d01
lb 2000xx0e001000000000f70732f893043d01
rb 2000xx0e002000000000f70732f893043d01
sync 2000xx0e010000000000f70732f893043d01
up 2000xx0e000100000000f70732f893043d01
down 2000xx0e000200000000f70732f893043d01
left 2000xx0e000400000000f70732f893043d01
right 2000xx0e000800000000f70732f893043d01
LS button 2000xx0e0040000000002406ddf31f123907
RS button 2000xx0e0080000000002406ddf31f123907
[left trigger] <right trigger> 20002d0e0000[5c02]<5901>f109e8f924086f03
rumble (I think) 0320xx0480010000

FranticRain, 2015-06-25:
Does the controller vibrate on start like in the Windows 7 driver? Or is that gone in Windows 10?

FranticRain, 2015-06-25:
Okay, huge thanks to @StillScreaming for getting all that info for me. It was incredibly useful. I think I've figured all of it out except the rumble. Let me know if this works at all: https://dl.dropboxusercontent.com/u/33409904/Xone-Driver-1.0.4a.dmg

StillScreaming, 2015-06-26:
Yeah it vibrates when you connect it in Windows 10

StillScreaming, 2015-06-26:
1.0.4a does the same thing as 1.0.4. Plug it in and it vibrates once(logo doesn't illuminate), shows up as Controller 1 in settings, load any game (Race The Sun, Geometry Wars 3, Lethal League, Nidhogg, Sportfriends), no response from controller

Sidenote: Headphone jack works in Windows10

KingSlayer816, 2015-06-26:
I can confirm that this ^ is the same case for Mac OSX 10.10

FranticRain, 2015-06-26:
Okay, I added a bunch of new packets. Maybe one of them will work. Check your console after you plug the controller in, and see if it says: "Xbox One Contoller - 2015 Edition."

https://dl.dropboxusercontent.com/u/33409904/Xone-Driver-1.0.4b.dmg

SmirksAlot, 2015-06-27:
i just tried the 1.0.4b driver and yes, the console says 2015 edition (twice actually), but the controller still rumbles and then nothing else works and no X lighting up.

FranticRain, 2015-06-27:
Okay, this one is literally every packet. Let me know if it works.

https://dl.dropboxusercontent.com/u/33409904/Xone-Driver-1.0.4c.dmg

SmirksAlot, 2015-06-27:
sorry man, no change with 1.0.4c

FranticRain, 2015-06-27:
I've officially run out of ideas. I made a program that will let you send bytes to the controller by hand. The format is: "XX XX XX XX XX" in Hex. So for example, if you were using it with the original Xbox One Controller (which you can't, its hard coded for the new one), you would enter: 05 20 to start the controller up. You can try your own packets, but I'm not 100% sure if it will even work. It works to start the original Xbox One controller, but I can't send a rumble packet at all. Hopefully you all can figure something out.
https://dl.dropboxusercontent.com/u/33409904/Xone%20Packet%20Tester.zip

FranticRain, 2015-06-29:
Okay, here is another build. I really hope this one works. https://dl.dropboxusercontent.com/u/33409904/Xone-Driver-1.0.4d.dmg

HoodedxSaints, 2015-07-10:
Unfortunately 1.0.4d one doesn't work for me either with my new covert forces controller

FranticRain, 2015-07-10:
That's crazy. I don't understand what the deal is. The packets are identical. I'll look for a deal on one of these controllers and then get it for myself, because something is really crazy here.

HoodedxSaints, 2015-07-10:
Awesome, I appreciate the effort to try to solve this problem. Anything I can do to help?

FranticRain, 2015-07-10:
If you have a Windows 10 machine, you could sniff the packets again like I've outlined above. Otherwise, all you can do is try and bribe someone at Amazon to put these controllers on sale on Prime Day.

HoodedxSaints, 2015-07-10:
Unfortunately I'm waiting for the official release of Windows 10, so I don't have it yet.

benzeng94, 2015-08-17:
Hi,  have this controller, by any chance have you gotten any progress?
Thanks

benzeng94, 2015-08-17:
Hi, by any chance do you have an update for this please?

joemsak, 2015-09-02:
FWIW I bought a normal XBOX One controller (with charge + play) today and it has the same vendor/product ID, and with the latest release found at this repo, I am having the same issues. Rumble on plug-in, no light up, no cooperation in system preferences pane. OS X 10.11 beta (latest developer release, 8th beta)

joemsak, 2015-09-02:
UPDATE: it works with the latest xbox one controller beta driver you mentioned in #71

FranticRain, 2015-09-02:
That controller you bought likely has a 3.5mm jack on the bottom of it. For some reason, Microsoft released the 2015 revision with Covert Forces first. Now all new controllers have that feature, and the new init packets. We are still working out the bugs and trying to get a full release done, but once we get it out, this repository will be deprecated. For anyone else looking to make these controllers work, uninstall my driver and [install this one instead.](https://github.com/d235j/360Controller/releases/tag/v0.15_beta2_unofficial) Watch that repository for future updates.

closed 2015-10-05 by FranticRain