Drewbadour/Xone-OSX #64: xone installed, controller recognize, pref pane solid, no input to CoD4

billyjacks, 2015-06-27:
Hi, thank you for your hard work FranticRain. I have the controller up and running, recognized, it's good to go. When I launch CoD4 (from Mac App Store) it doesn't seem to accept any input from it. I've been searching everywhere online and can't find a solution. Any ideas? Thanks

FranticRain, 2015-06-27:
Hey, the best course of action is probably to uninstall my driver, and install this one instead: https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview7.pkg

I have ported my Xbox One controller code into that driver. It is a revived version of the Tattiebogle driver that people used for the Xbox 360 controller on OS X. CoD4 is a game that was brought to OS X by Aspyr, and Aspyr stinks at controller support. In most of their games, they hard-coded the games to ONLY support the Xbox 360 controller. Installing a driver would actually break their support for the controller! However, as they did more and more work, they got marginally better, and started using the Tattiebogle driver AND ONLY the Tattiebogle driver to get controller inputs. So this other driver SHOULD work with Call of Duty 4, but I can make no promises. Let me know how it works.

billyjacks, 2015-06-27:
You're a boss. Thanks for the response. The new driver you just gave me works perfectly as well, but CoD4 still isn't picking it up. I'm trying to find a setting within the game such as "Switch to Controller" or anything...but all it does is pick up KB/M... : /

FranticRain, 2015-06-27:
Yeah, it looks like it was just a bad Aspyr port job. There is one way to fix it, and that is to map the controller keys on your keyboard and/or mouse. Its a feature that I've been wanting to add to the 360Controller driver for a while, but I need them to fix a settings bug before I can. That is probably a ways off, so don't wait for me on that one. In the meantime, if you are willing to pay some cash, [JoystickMapper](http://joystickmapper.com/) will let you map the controller so that you can play. It even has a handy little video that should show you how to set it up for CoD4, no problem!

billyjacks, 2015-06-27:
Ok I got Joystick Mapper (your app can do this too though, no?)...the binding seems to be working but still nothing picked up by CoD...I'll keep trying. Thanks for your help Frantic

FranticRain, 2015-06-27:
No, my driver cannot bind to keyboard values. It is bound to traditional controller inputs. If Joystick Mapper really is working, then it should be putting out keyboard values. I.E. W, A, S or D for the left stick. I assume you can control the game with your computer keyboard, so I do not understand why Joystick Mapper is not working. You could make sure that joystick mapper is outputting keyboard values by opening up TextEdit and spinning the left stick around.

billyjacks, 2015-06-27:
Tried TextEdit...nothing. At this point it has to be some kind of OS X issue. It's frustrating because everything seems to be working, there's definitely input being capture and bound, it's just not executing the bound key.

billyjacks, 2015-06-27:
Thanks again for your help by the way, this is crazy that you're helping me out so quickly

FranticRain, 2015-06-27:
Have you updated Joystick Mapper? There was bug in versions earlier than 1.1.2 that caused it to not work on 10.9.

billyjacks, 2015-06-27:
[image]
[image]

FranticRain, 2015-06-27:
Could you try the "Example: FPS" preset for Joystick Mapper, and see if it work? If it doesn't, I'll get Rodrigo, the maker of Joystick Mapper in here to help you out.

billyjacks, 2015-06-27:
Well I feel stupid. I just needed to activate the checkbox for the damn thing. That should be the first trouble shooting step on the site. wow. *facepalm* Thanks man....appreciate the help. have a good night.

FranticRain, 2015-06-27:
Glad it works for you!

closed 2015-06-27 by FranticRain