Drewbadour/Xone-OSX #65: Dungeon Defenders (Steam)

Pings, 2015-07-03:
Seems to almost work, Y axis on left stick is sometimes inverted..? and click left and right stick are swapped with back and start. May just be Dungeon Defenders and their garbage Mac port but thought I'd let you know. 

Works great with Steam streaming to windows PC though!

FranticRain, 2015-07-03:
When you go to the rebinding menu within Dungeon Defenders and press start,
back, right stick click and left stick click, do they show up as the
correct buttons there? As for the occasional incorrect sticks, that sounds
like a game level problem.

Additionally, you might be better off if you use the revived Tattiebogle
driver. Games tend to like that one a bit better. I ported all of my Xbox
One code to it, so it works great. You will need to uninstall my driver
first, and then install the new one. That driver is available here:

FranticRain, 2015-07-03:
Email stripped the link out, lets try again: https
<https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview7.pkg>
://
<https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview7.pkg>
dl.dropboxusercontent.com
<https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview7.pkg>
/u/14890209/
<https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview7.pkg>
Install360Controller0.15Preview7
<https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview7.pkg>
.pkg
<https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview7.pkg>

If that doesn't work, it's on the releases page in bold for third party
controllers.

Pings, 2015-07-03:
When I try remap the buttons all respond but are messed up but the sticks don't give any input. I'll try that new driver thanks!