Drewbadour/Xone-OSX #66: XboxOneController2015 not working

clee, 2015-07-13:
It detects it in the system preferences pane (I see 'Controller 1' in the dropdown), but none of the buttons show activity. Using 1.0.4.

FranticRain, 2015-07-13:
This issue is already covered in #63. This is a known issue caused by a change in the packet used to start the controller. The new controller doesn't start the same way the old one did You can tell that this is the issue because the Xbox logo doesn't light up. I'm going to have to purchase one for myself to solve the problem. I'm waiting for a good deal though, because I already have too many of these controllers.

FranticRain, 2015-07-13:
Actually, if you could do me a favor and test something for me. Uninstall my driver, and [go to this website](http://www.html5rocks.com/en/tutorials/doodles/gamepad/gamepad-tester/tester.html). Press some buttons on the controller and see if the controller starts up as expected. If it does, then that's a nice tip. If it doesn't, then no harm no foul.

clee, 2015-07-13:
Okay, will do!

clee, 2015-07-13:
No luck. It vibrates when I connect the microUSB cable, but the Xbox light doesn't turn on. Mashing all of the buttons and triggers (and pressing them carefully one by one) does not seem to do anything on that page.

FranticRain, 2015-07-13:
Yeah, thats what I expected. You'll just have to wait until I get my hands on the new controller.

clee, 2015-07-13:
Fair enough! I still have my Day One controller so I can use that on my computer for now. Thanks for the quick responses!

FranticRain, 2015-10-05:
I completely forgot about this issue, but the 2015 controller is working now. It didn't play nicely with my driver, but if you uninstall my driver and [install this one instead](https://github.com/360Controller/360Controller/releases), it will work perfectly!

closed 2015-10-05 by FranticRain

clee, 2015-10-05:
Thanks for the update! I had already uninstalled this driver and installed the other one, but I think the explanation will be useful for anybody who finds this via Google in the future :)