Drewbadour/Xone-OSX #67: Not compatible with El Capitan

blindpilot17, 2015-07-17:
Hey I know it hasn't been released yet but just to make aware that this driver isn't working on the next operating system for Mac.

FranticRain, 2015-07-17:
Could you uninstall my driver and try out this driver: https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview7.pkg

Once they finally put out an official release, this driver will be deprecated. Hopefully that will happen before El Capitan releases. I haven't read any news about kext changes in El Capitan. Will have to look into that.

blindpilot17, 2015-07-18:
it works!! Thanks a ton!!

FranticRain, 2015-07-18:
Awesome. Hopefully everything works out and then I will deprecate this driver at the proper time.

closed 2015-07-18 by FranticRain