Drewbadour/Xone-OSX #68: Controller not working on steam

lplplplp, 2015-07-30:
When I try to use the controler in big picture mode it does not reconize the controller

FranticRain, 2015-07-30:
Did you bind the controls in the settings menu? Also, check if the controller is found by the preference pane application.

lplplplp, 2015-07-30:
the controller works fine when i am previewing it but trying to use it in big picture mode does not work.

Standard xbone controller 
yomesite

FranticRain, 2015-07-30:
Within big picture, there is a settings menu with a controller option. Within that menu, you must bind the controller buttons to Steam Big Picture's bindings. Do that, and the controller will work.

lplplplp, 2015-07-30:
[image]
this is exactly what i see

FranticRain, 2015-07-30:
Okay, uninstall my driver using the uninstall.command included in the .dmg that you downloaded, and install [this driver instead.](https://github.com/d235j/360Controller/releases/tag/v0.15_beta_unofficial) I moved all of my Xbox One controller code into it, and its much better supported. It's still weird that this is an error, because it works just fine for plenty of people in the same situation.

lplplplp, 2015-07-30:
I just really confused and new to this can you add me on skype and we will talk there my skype is sam.schorer

lplplplp, 2015-07-30:
i can talk in the moring

FranticRain, 2015-07-30:
Just double click the "uninstall.command" file included in your Xone-Driver-1.0.4.dmg. Then, download "360ControllerInstall_0.15_beta_unofficial.dmg" from the link I sent you, and install that. However much I would like to help you personally, I do not have the time to do so on a free project that I do in my spare time.