Drewbadour/Xone-OSX #69: Xone-OSX conflicts with d235j/360controller in some programs

blakekl, 2015-08-02:
i use the d235j/360Controller driver for my 360 controllers and I'm trying to get this controller to work with it. Problem is that sometimes I use ControllerMate to add joystick support to games that don't support it natively. It see's these two devices as the same, but their button layouts are completely different. This makes working with both impossible in my situation. The devices appear to be different in my system profiler, but the same in controllermate. Interestingly, controllmate see's the device as a different device when I use the 'OLDBIND' version, but the buttons are just wonky in that case.

Below are screenshots of the system profiler for both a 360 and xone controller. Also, the bindings shown for 360 controller in ControllerMate. I'll add the xone bindings in controllerate after installing that driver again.

System Profile of 360 Controller
[image]

System Profile of XOne Controller
[image]

Controllermate Bindings of 360 Controller
[image]

Controllermate Bindings of XOne (OLDBINDINGS version)
[image]

Controllermate Bindings of 360 and normal XOne version (XOne input on left, perceived 360 output on right)
A = RStick Button
B = LB
X = RB
Y = Guide Button
LB = Start
RB = Back
Menu = DPad Up
View = DPad Down 
Guide = Nothing?
DPad Up = Nothing?
DPad Down = Nothing?
DPad Right = Byte Count?
DPad Left = Nothing?
All Sticks = A
LT = A + Y
RT = A

So pushing Y on my XOne button registers as the guide button on a 360 controller in controller mate.
This means I'm unable to use a 360 controller and an xbox one controller at the same time because the bindings are so off. Not sure what the issue is here, but does it have something to with the changes to make older games that have built in support for the 360 controller work correctly?

FranticRain, 2015-08-02:
The latest version of the d235j/360controller driver includes Xbox One controller support that I added. Just uninstall my driver, and update 360Controller, that will most likely solve most of your problems.

blakekl, 2015-08-02:
Alright, will give that a try.

blakekl, 2015-08-02:
Excellent. That solves my problem perfectly. Thanks for the quick response too. Greatly appreciated.

closed 2015-08-02 by blakekl