Drewbadour/Xone-OSX #71: Newer xbox one controller not working.

Turtlecat33, 2015-08-19:
I have a good cord and a newer controller, (it has the extra headphone jack on the bottom) but when I try to connect it it doesnt pick up the controller. Help pls

mattwagstaff, 2015-08-19:
Same issue here I think... .looks like it's on the cards to fix.
#63

FranticRain, 2015-08-19:
That controller will not work, and will never work with this driver. However, it will work with the [360Controller repository](https://github.com/d235j/360Controller/) as soon as [they accept my pull request.](https://github.com/d235j/360Controller/pull/120)

Turtlecat33, 2015-08-19:
Okay! Im using OpenEmu to play emulators on my macbook pro. Any recommendations on a controller to get for it?

geowar1, 2015-08-19:
PS3 & PS4 controllers will work “out of the box”.

> 
> Okay! Im using OpenEmu to play emulators on my macbook pro. Any recommendations on a controller to get for it?

FranticRain, 2015-08-19:
This driver will be deprecated when there is a full release of the 360Controller driver, so feel free to use the Xbox One controller. It works really well. OpenEmu uses a really great controller library called DDHidLib. It will work with pretty much all controllers. Even stuff like usb SNES controllers, if you're into that kind of thing.

closed 2015-10-05 by FranticRain