Drewbadour/Xone-OSX #72: Xbox One Controller not working with FIFA 14

NeoStargate, 2015-08-21:
I am using CrossOver, but I think it is FIFA 14 the one that doesn't support the Xbox One controller. While the controller is being detected, the button mapping is all wrong. However, there are two files that seem to handle button mapping:

http://pastebin.com/iiEeHWX6
http://pastebin.com/QaFU9vJH

I wonder if by modifying these I can get the controller to be properly recognised.

FranticRain, 2015-08-21:
The best thing to do is uninstall my driver and install [this one instead.](https://github.com/d235j/360Controller/releases/tag/v0.15_beta_unofficial) I've been moving my code into it, and games support it much better than they do mine. It should fix your problem perfectly.

NeoStargate, 2015-08-24:
Hello,

I tried it, but the problem persists.

FranticRain, 2015-08-25:
If it doesn't work with that driver, then it probably won't work at all. Sorry.

closed 2015-10-05 by FranticRain