Drewbadour/Xone-OSX #74: Xbox ONE mini Support?

brykoch, 2015-09-07:
Mac recognizes the controller in the USB device tree. Any insight on getting it working? Here's the info displayed:

Xbox ONE mini:

  Product ID:	0x541a
  Vendor ID:	0x24c6
  Version:	1.01
  Serial Number:	0000F531B56163EB
  Speed:	Up to 12 Mb/sec
  Manufacturer:	BDA
  Location ID:	0x14500000 / 9
  Current Available (mA):	500
  Current Required (mA):	Unknown (Device has not been configured)

FranticRain, 2015-09-07:
As the README says, this driver does not support third party controllers. Uninstall my driver and [install this one instead.](https://github.com/d235j/360Controller/releases/tag/v0.15_beta2_unofficial) I have been moving my code into that driver because it works much better than mine.

closed 2015-09-07 by FranticRain

brykoch, 2015-09-07:
Thanks, FranticRain. I meant to kill this post instead of posting… Came across your other posts about the topic and accidentally posted when I came back to this tab.

Appreciate the quick reply.

FranticRain, 2015-09-07:
No problem.