Drewbadour/Xone-OSX #76: X-Box Controller Not Working on Steam

SirTankshire, 2015-09-27:
The latest version will not work when running Fall Out New Vegas in Steam. Running Mac OS 10.10.5.

Any help please.

FranticRain, 2015-09-28:
Uninstall my driver, and install [this one instead.](https://github.com/360Controller/360Controller/releases/tag/v0.15_beta3_unofficial) I have been moving my code into it because it has much better game support. As soon as they put out and official release of their driver, this one will be deprecated.

Is this the first game you've tried using with this controller? Did it work before with some other game? Let me know if you have any success with the 360Controller driver.

SirTankshire, 2015-09-28:
Thanks for the quick reply. This is the first game. Its still not working. I did try the XBOX One Controller enabler. Is there an issue in having installed this first?

Thanks

FranticRain, 2015-09-28:
No. That is a program and it should have been fully removed upon deletion. Now, can you follow the debug steps listed in the README? Does the controller show up in the list of USB devices?