Drewbadour/Xone-OSX #84: About Button Layout

huaikitty, 2015-12-03:
About Button Layout

How to edit Button Layout?

Xbox one controller

example "Controller Button A" is  not "HID Button 1",however is "HID Button 11",Why? How to fixed it,Thanks.
[image]

FranticRain, 2015-12-03:
This driver is deprecated. Please use [this driver](https://github.com/360Controller/360Controller/releases) instead.
Unfortunately, this driver only works with the original Xbox One controller for some reason, so I've moved all of my code to a more robust driver that supports later Xbox One controllers in addition to 3rd party controllers.

The 360Controller repository supports remapping.

closed 2015-12-03 by FranticRain

huaikitty, 2015-12-03:
Thanks,I try it.